SpacehuhnTech/esp8266_deauther #622: Wipe EEPROM settings?

poorredneck, 2017-09-26:
I have a Wemos Mini D1 that I had a much much older version of deauther on. I used the NodeMCU firmware programmer to flash the 1MB version, however it seems the old settings are still in the EEPROM and are causing issues with stability. The SSID and password are still what was set a long time ago, and are not erasing when I flash the new v1.6. I can connect to the wemos mini running 1.6, and load the splash screen, but after that it just crashes. 

I realize this isn't necessarily a deauther specific issue, but I can not figure out how to wipe the EEPROM on the wemos mini d1. 

I appreciate any assistance.

kt20062011, 2017-09-26:
http://www.pratikpanda.com/completely-format-erase-esp8266-flash-memory/.  I asked for this method and everything worked

jLynx, 2018-01-05:
Take a look at the FAQ

closed 2018-01-05 by jLynx