SpacehuhnTech/esp8266_deauther #624: How do i view the attack log?

crystalg21, 2017-09-27:
What command do i use in Arduino serial monitor at 115200 baud to view the log of all the attacks that happened? Thanks

tobozo, 2017-09-29:
There is no such implementation yet but it's not that far.

```
    void loadLogFile() {
        File logFile = SPIFFS.open("/log.txt", "r");
        server.streamFile(logFile, "text/plain");
        logFile.close();
    }

    (...)

    void setup() {

        (...)

        server.on("/log.txt",loadLogFile);

        (...)

    }

```

this will probably be added soon to the testing branch, please leave this ticket open until then

tobozo, 2017-10-16:
closing this now it is merged into the testing branch

closed 2017-10-16 by tobozo