SpacehuhnTech/esp8266_deauther #629: Can`t connect to 192.168.4.1 Help pls

TheVados, 2017-10-04:
My phone does not connect to 192.168.4.1. While other devices are connecting, help please

tobozo, 2017-10-04:
Pasting this here so you can't "accidentally" delete it while redacting your question:

````
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware/Flashing:
Board:				?ESP8266 dev module?Tindie OLED?Wemos Mini?Other (specify)?
Using OLED:   ?YES?NO?
Using WebServer: ?YES?NO?
Using Serial: ?YES?NO?
Installation mode:  ?binary?source?
Code/binary version: ?1.5?1.6?
Flash size:  ?1M/512K?
Flash method:							?esptool?Arduino IDE?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?
Powered by:  ?USB?Battery?

### Description:
Describe your problem here, provide as many details as possible.
Make sure you read the FAQ first, incomplete bug reports will be CLOSED without notice.

What you're expecting:  ---
What you get instead: ---

### Debug Messages:
```
Put the serial output here 
````

tobozo, 2017-10-09:
Closing this ticket, feel free to reopen when you can qualify the issue with more details (so we can help, not assist).

closed 2017-10-09 by tobozo