SpacehuhnTech/esp8266_deauther #633: How can I update a old version?

Sleek-Tech, 2017-10-08:
How can I update the deauther? It won't scan for wifi access points anymore. Thanks

Phoenix1747, 2017-10-08:
You have to flash a new release if you want to update the software itself.

dfghdsg, 2017-10-12:
Hi, I saw a video about  WiFi Jammer... wanted to repeat but it did not work ... I'm new to this :)
[image]
[image]
[image]
[image]
[image]

tobozo, 2017-10-12:
@ dfghdsg  if you just happened to read the documentation clearly visible on the screenshot you just posted, you know it's not a jammer.

closed 2017-10-12 by tobozo

dfghdsg, 2017-10-12:
I did everything on this video (not advertising), but there is not that old sketch ...  https://www.youtube.com/watch?v=RZt2IBZ7t_4

tobozo, 2017-10-12:
I'm sure you'll find a way to get it working, it's all about trial and error,
Just have faith in the ReadME because it gets updated while the video does not.

dfghdsg, 2017-10-12:
Thank you very much for the support and advice ... I will try again :)