SpacehuhnTech/esp8266_deauther #636: crash, loosing connection

s199, 2017-10-12:
Hello 
i flashed an esp with the latest release but the only page i can access is the main page with scan (very nice new ui!), i can scan and select a network but then i cant access any other page, sometimes i loose connection directly, sometimes the page open but stay white with some text (credit and github adress).
i have another esp flashed with an older bin and it work fine :/

the esp is hooked to my mobile phone (huawei p8), im using the oled but no button (still need to add them). Power do not seems to be the problem as i tested also with a 5v 2.5a usb (same problem)

### Hardware/Flashing:
Board:	esp8266mod lolin
Using OLED:   yes but no button
Using WebServer: hu
Using Serial: dont think so
Installation mode:  bin
Code/binary version: 1.6
Flash size:  1M
Flash method:	flasher				
Flash Frequency: dunno
Upload Speed:						115200
Powered by:  huawei p8 or usb via charger 5v 2.5a
browser : chrome (android 6.0)

any help is welcome :) \O/
++

tobozo, 2017-10-13:
Interferences maybe ? I found out that when my phone is too close to the ESP it doesn't behave the same.

s199, 2017-10-13:
i dont think so because i tried with a usb charger instead of the phone also and got the same probleme, im loosing connection as soon as i click to any other page (setting, attack, etc) :/

tobozo, 2017-10-13:
Can you try to erase_flash and reflash after ?

Also if you're compiling from source with Arduino IDE there are some different settings you can try in the board menu.

Actually the serial output you'll get when the failure happens can be helpful.

Note that the ESP lolin uses a CH340 serial adapter.

s199, 2017-10-13:
i flashed using the BIN file, i will try to add the three buttons in a moment (3dprinting case now) and see if the esp work with button only first, if it do not work i will reflash another brand new esp and see whats going on. in the meantime i will try to figure out how to use the serial output :)

s199, 2017-10-13:
ok i've added the button and everything works fine, powered by the smartphone, i can access all pages, come back to main, launch attack etc.
its the same unit as previously, with same bin.
Looks like it the mobile side which cause problems :/
i have no clue at all how to get a log or some error file, if i can help in any way tracking the issue jut lemme know :)

tobozo, 2017-10-13:
this is good news :-)

If you're curious about using the serial console, it's just a matter of having the drivers installed (CH340 for Windows or OSx) and having the Arduino IDE running while the ESP is powered via your USB. It won't be enough to compile & upload the source but it'll give you a serial console where you can eventually spot errors, good for debugging if you encounter another anomaly.

Your phone USB has a limited output power though. Even if it provides enough for the ESP to boot, it's highly doubtful it will provide enough energy for the ESP to handle the WiFi and the webserver.

s199, 2017-10-13:
im going to try the serial thingy tonight so i can post the log.

regarding the usb power from phone, i did try with a 5v 2.5a usb (from battery charger instead of phone to power the esp) and the result was exactly the same.
So esp hooked to external usb 5v 2.5a and phone only to get the interface, same behaviour, only page i can access was main page, then i loose connection everytime. (i rebooted phone, made sure there no app running, cleared cache etc).

tobozo, 2017-10-13:
Some chargers can deliver 2.5A but only after a negotiation with the device (especially iPhone chargers), so you may have a module that *can* send 2.5A but is stuck a 250mA or so because no negotiation occurred. This could explain the "same behaviour on phone and on charger".

If you can't verify the power with a multimeter, the only way to be sure it's not power related is to try with a battery (i.e 18650 battery with a power booster) or use a real power module, not a charge module.

s199, 2017-10-13:
haa ok, ill get a multimeter and post results alongside the error logs from serial later, its a 18650 charger with usb out so maybe usb has less power :) ill check this out asap!

s199, 2017-10-14:
Ola, i tried the serial log but i think it wont help much, the log stops when i loose connection :/// it doesnt seems to list any reason.
here is what i did : esp hooked to laptop, arduino running, i start the app on the phone, get to main page, perform a scan and then as soon as i click another page i loose connection :/
here is the log

SSID: pwned

SSID length: 5

SSID hidden: 0

password: deauther

password length: 8

channel: 1

Default MAC AP: 2e:3a:e8:08:e4:f7

Saved MAC AP: 2e:3a:e8:08:e4:f7

MAC AP random: 0

Scan hidden APs: 1

deauth reson: 1

attack timeout: 300

attack packet rate: 10

client scan time: 15

use built-in LED: 1

channel hopping: 0

multiple APs: 0

multiple Attacks: 0

mac change interval: 4

1s beacon interval: 0

LED Pin: 2

Trying to sync the MAC addr with settings

Synced with saved mac addr : 2e:3a:e8:08:e4:f7

nameList cleared

Attack LED : OFF

Starting WiFi AP:

SSID     : 'pwned'

Password : 'deauther'

-----------------------------------------------

 generating MACs...done

Starting...

sending AP scan result JSON done

starting AP scan...

MAC - Ch - RSSI - Encrypt. - SSID - Hidden

Scan results: 9

0 - e0:b9:e5:14:c9:d7 - 1 - -53 - WPA2 - Derfworld - 0

1 - b8:d9:4d:bd:83:16 - 1 - -78 - WPA2 - Petit_prince - 0

2 - 30:d3:2d:7c:d2:7e - 6 - -74 - WPA2 - WiFi-2.4-43AB - 0

3 - 9c:97:26:99:4d:55 - 6 - -90 - WPA2 - WiFi-2.4-4D55 - 0

4 - 9e:97:26:99:4d:58 - 6 - -92 - none - PROXIMUS_AUTO_FON - 0

5 - a0:21:b7:d8:0d:39 - 11 - -93 - WPA* - VOO-036272 - 0

6 - 9e:97:26:25:fe:50 - 11 - -94 - none - PROXIMUS_AUTO_FON - 0

7 - a2:21:b7:d8:0d:3a - 11 - -93 - none - VOO_HOMESPOT - 0

8 - a2:21:b7:d8:0d:3b - 11 - -93 - none -  - 1

scan done

sorting APs 

--------------

-53 < -78

switched: -78 > -74

-78 < -90

-90 < -92

-92 < -93

-93 < -94

switched: -94 > -93

switched: -94 > -93

--------------

-53 < -74

-74 < -78

-78 < -90

-90 < -92

-92 < -93

-93 < -93

-93 < -93

--------------

-53 < -74

-74 < -78

-78 < -90

-90 < -92

-92 < -93

-93 < -93

--------------

-53 < -74

-74 < -78

-78 < -90

-90 < -92

-92 < -93

--------------

-53 < -74

-74 < -78

-78 < -90

-90 < -92

--------------

-53 < -74

-74 < -78

-78 < -90

--------------

-53 < -74

-74 < -78

--------------

-53 < -74

sending AP scan result JSON .........done

tobozo, 2017-10-14:
I edited your post for readability :p 

The connection loss happens on my DSTIKE build too (using the `testing` branch) so we know it's not hardware related, and the latest changes don't solve it.

I may have a lead on something but I need to do some testing first and I only have two different builds available here, would you like to try to setup a full Arduino environment so you can help me with it?

If you've managed to install the IDE and connect the ESP to the Serial already, it's only a matter of carefully following the steps in the [ReadME](https://github.com/spacehuhn/esp8266_deauther/blob/testing/README.md) and cloning the [testing](https://github.com/spacehuhn/esp8266_deauther/tree/testing) branch.

s199, 2017-10-14:
i followed the procedure as described, flashed the ino on the testing page. (had a little grafic while flashing, version still says 1.6 and there is an awesome screensaver not sure it was there before).
for some reason i cant access anything from the UI now, i tried on mobile and laptop (not even the main page). i can still use the esp with the screen and button, its not frozen.

tobozo, 2017-10-14:
Did you install and use the [data uploader](https://github.com/esp8266/arduino-esp8266fs-plugin)?

s199, 2017-10-14:
yes, everything installed

tobozo, 2017-10-14:
Sounds like SPIFFS is empty do you see SPIFFS errors in the serial console while this happens?

Also there's a workaround for low memory profiles you may want to try.
You can free some memory by commenting out the vendor list.

The "oui.h" file has 23k lines of optional data stored in PROGMEM, maybe if you comment out the block of data between this:

```
#ifndef oui_h
#define oui_h
/*
  Based on Wireshark manufacturer database
  source: https://www.wireshark.org/tools/oui-lookup.html
  Wireshark is released under the GNU General Public License version 2
*/

const static uint8_t data_vendors[] PROGMEM = {
```

and this:
```
};
#endif
```

s199, 2017-10-14:
generating MACs...done
Starting...
[SPIFFS FAIL]
[SPIFFS FAIL]

ill have a look at the oui.h file now

s199, 2017-10-14:
seems commented out already ? :/

tobozo, 2017-10-14:
it seems commented out but it isn't.

From the SPIFFS error messages you got in the serial console, it seems the sketch data upload didn't work.

Make sure the project is open in the Arduino IDE, the correct port is selected and no serial console is open before you run the utility from `Tools/ESP8266 Sketch Data Upload` menu again, also power cycle the ESP afterwards.

s199, 2017-10-14:
i think it worked now, i pasted the serial on pastebin 
https://pastebin.com​/47c9nDpg

tobozo, 2017-10-14:
There's [this plugin](https://github.com/me-no-dev/EspExceptionDecoder) to help decode exceptions produced by the ESP. I tried to paste your exception code in my decoder but all I can see is that you're using a Mac ;p

Does the [SPIFFS FAIL] and the core dump happen together ? Is it after you access the web UI ?

s199, 2017-10-14:
i dont have anything from apple here :) its an huawei phone and an hp laptop.
everything happen together when i type 192.168.4.1, i loose connection directly

tobozo, 2017-10-15:
Can you try insert this somewhere in your setup() function) and see what it outputs ?

``
if(SPIFFS.format()) {
  Serial.println("SPIFFS Formatted, remove this code block and use ESP8266 Sketch Data Upload");
} else {
  Serial.println("SPIFFS Format failed, check your Flash Size board settings!");
}
```

After this code runs successfully, you'll be in a situation where nothing is on the SPIFFS. 
Connect your phone and compare the symptoms with the previous situation; and if they match you'll have confirmation the sketch data uploader failed.

Once you have captured the serial output for this test and identified the situation, you may remove the SPIFFS.format() code block and proceed to re-upload the /data folder using the ESP8266 Sketch Data Upload from the tools menu (plaese paste the output of the Arduino log window here).

Depending on the result, next adjustment could be some tweaking on the board choice/settings.
Actually what board settings are you using in your Arduino IDE ?

For my DSTIKE build I use this:
  Board: Generic ESP8266
  CPU Freq: 80MHz
  Flash Size: 1M(256K SPIFFS)

You may have to try different values for Flash Size, I tried them all before finding out 1M(256k) was stable with my build, your mileage may vary.

Note that every time you change your board settings from the menu, you must re-flash the ESP (without the SPIFFS.format code block) && re-use the ESP8266 Sketch Data Upload && power cycle the ESP.

s199, 2017-10-15:
Hello Tobozo, first lemme thank you for your time, i learned more in two days than in a year browsing youtube. i tried the procedure but its a no go, still got spiff fail etc. Im a bit lost in all im doing to be honest so its probably an error on my side :/ i think i will reinstall everything from scratch, all lib etc i feel there is something wrong with my installation.
my esp8266 with led and button works super fine, ill use it instead of the phone for now, and retry to install everything properly once i got a clean installation :) (might take a few days)

tobozo, 2018-03-24:
probably fixed in the next version, closing this

closed 2018-03-24 by tobozo