SpacehuhnTech/esp8266_deauther #646: Windows 10 (pre-new developer build) BSOD.

kres0345, 2017-10-27:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware/Flashing:
Board:				?Esp8266 nodemcu  0.9 I think. 
Tindie OLED?Wemos Mini?Other (specify)?
Using OLED:   ?YES?NO?
Using WebServer: ?YES?NO?
Using Serial: ?YES?NO?
Installation mode:  ?binary?source?
Code/binary version: ?1.5?1.6?
Flash size:  ?1M/512K?
Flash method:							?esptool?Arduino IDE?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?
Powered by:  ?USB?Battery?

### Description:
I am using the largest bin no oled just the direct purchase from eBay. The problem triggers when unplugging the esp8266/nodemcu, the problem is BSOD. I haven't modified anything. And it's windows 10(pre-new developer update).

What you're expecting:  ---
What you get instead: ---

### Debug Messages:
```
Put the serial output here

kres0345, 2017-10-27:
PS followed seytonic update

tobozo, 2017-10-27:
Are you using the [official Wemos drivers](https://wiki.wemos.cc/downloads) ? 

Also if you can get your hands on another OS and try to reproduce the problem, does it still give you a BSOD?

kres0345, 2017-10-27:
I havent heard about the Wemos drivers and no I only have this pc. Well my android phone, but the problem havent occured there yet

tobozo, 2017-10-27:
I hope you never get BSOD on android :-)

Can you try to remove your actual Windows driver + install the one from the Wemos link and tell us if you still get the BSOD ?

kres0345, 2017-10-28:
I haven't experienced the BSOD in a long time so it isn't currently a problem. I Just wanted to let you know, there at lease were a bug :-D. And I hope too I won't get a Blue Screen of Death on my android XD
And I installed the drivers the same time you posted the link.

tobozo, 2017-10-28:
closing this issue, feel free to reopen if the symptoms are back

closed 2017-10-28 by tobozo