SpacehuhnTech/esp8266_deauther #649: error: espcomm_open failed

m00n21, 2017-11-02:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware/Flashing:
Board:NodeMcu Lua CH340G ESP8266			?ESP8266 dev module?Tindie OLED?Wemos Mini?Other (specify)?
Using OLED:   ?YES?NO?no
Using WebServer: ?YES?NO?no
Using Serial: ?YES?NO?no
Installation mode:  ?binary?source?source
Code/binary version: ?1.5?1.6?
Flash size:  ?1M/512K?i dont know
Flash method:		i tryed both and didnt work					?esptool?Arduino IDE?
Flash Frequency:		40			?40Mhz?
Upload Speed:			115200			?115200?
Powered by:  ?USB?Battery? mico cabele

### Description:
Describe your problem here, provide as many details as possible.
Make sure you read the FAQ first, incomplete bug reports will be CLOSED without notice.

What you're expecting:  ---  i click up load arfter all of the steps and it comes up with an error 
What you get instead: --- error: espcomm_open failed

### Debug Messages:
``` error: espcomm_open failed
Put the serial output here

bolbolmm, 2017-11-03:
go to Tools>Port>COM4 
and i think it will work

m00n21, 2017-11-05:
I only have COM1 an COM8

tobozo, 2017-11-05:
@m00n21 this is progress! 
You'll find out this com number will change every time you plug or unplug your ESP, so make sure you check this before flashing.

m00n21, 2017-11-05:
you fucking retards ,i know what port it is

closed 2017-11-05 by tobozo