SpacehuhnTech/esp8266_deauther #651: fail to flash

statix83, 2017-11-07:
### Hardware/Flashing:
Board:	Tindie OLED
Using OLED:   YES
Using WebServer: YES
Using Serial: 
Installation mode:  source
Code/binary version:1.6
Flash size:  512K
Flash method:	Arduino IDE
Flash Frequency: 160MHZ
Upload Speed: 115200
Powered by:  USB

### Description:
I am trying to change my ssid and password .... but i end up just changing my ssid  
but when i click save , it said error ... so i pres save a few times but keep saying error ... 
after i restart the deauther , the ssid changed but the password does not work 

i still can use the deauther from the oled display but i cant  connect to the webpage 

so i am trying to reflash the whole deauther ... 

i have followed all steps from the tutorial , but when i am trying to upload the file ... 
i got this error 

Arduino: 1.8.2 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 160 MHz, 115200, 4M (3M SPIFFS)"

Build options changed, rebuilding all
In file included from C:\Users\Runner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\ESP8266WiFiGeneric.h:26:0,

                 from C:\Users\Runner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:26:

C:\Users\Runner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\ESP8266WiFiType.h:35:14: error: multiple definition of 'enum WiFiMode'

 typedef enum WiFiMode 

              ^

In file included from C:\Users\Runner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:25:0:

C:\Users\Runner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\ESP8266WiFi.h:39:6: error: previous definition here

 enum WiFiMode { WIFI_OFF = 0, WIFI_STA = 1, WIFI_AP = 2, WIFI_AP_STA = 3 };

      ^

In file included from C:\Users\Runner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\ESP8266WiFiGeneric.h:26:0,

                 from C:\Users\Runner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:26:

C:\Users\Runner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\ESP8266WiFiType.h:38:13: error: invalid type in declaration before ';' token

 } WiFiMode_t;

             ^

C:\Users\Runner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp: In member function 'bool ESP8266WiFiAPClass::softAP(const char*, const char*, int, int)':

C:\Users\Runner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:86:14: error: 'class ESP8266WiFiClass' has no member named 'enableAP'

     if(!WiFi.enableAP(true)) {

              ^

C:\Users\Runner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:88:45: error: 'DEBUG_WIFI' was not declared in this scope

         DEBUG_WIFI("[AP] enableAP failed!\n");

                                             ^

C:\Users\Runner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:94:54: error: 'DEBUG_WIFI' was not declared in this scope

         DEBUG_WIFI("[AP] SSID too long or missing!\n");

                                                      ^

C:\Users\Runner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:100:62: error: 'DEBUG_WIFI' was not declared in this scope

         DEBUG_WIFI("[AP] fail passphrase to long or short!\n");

                                                              ^

In file included from C:\Users\Runner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:25:0:

C:\Users\Runner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\ESP8266WiFi.h:390:10: error: 'bool ESP8266WiFiClass::_persistent' is protected

     bool _persistent;

          ^

C:\Users\Runner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:127:17: error: within this context

         if(WiFi._persistent) {

                 ^

C:\Users\Runner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:135:51: error: 'DEBUG_WIFI' was not declared in this scope

             DEBUG_WIFI("[AP] set_config failed!\n");

                                                   ^

C:\Users\Runner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:140:52: error: 'DEBUG_WIFI' was not declared in this scope

         DEBUG_WIFI("[AP] softap config unchanged\n");

                                                    ^

C:\Users\Runner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:144:58: error: 'DEBUG_WIFI' was not declared in this scope

         DEBUG_WIFI("[AP] DHCP not started, starting...\n");

                                                          ^

C:\Users\Runner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:156:63: error: 'DEBUG_WIFI' was not declared in this scope

             DEBUG_WIFI("[AP] IP config Invalid resetting...\n");

                                                               ^

C:\Users\Runner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:165:53: error: 'DEBUG_WIFI' was not declared in this scope

         DEBUG_WIFI("[AP] wifi_get_ip_info failed!\n");

                                                     ^

C:\Users\Runner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp: In member function 'bool ESP8266WiFiAPClass::softAPConfig(IPAddress, IPAddress, IPAddress)':

C:\Users\Runner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:180:150: error: 'DEBUG_WIFI' was not declared in this scope

     DEBUG_WIFI("[APConfig] local_ip: %s gateway: %s subnet: %s\n", local_ip.toString().c_str(), gateway.toString().c_str(), subnet.toString().c_str());

                                                                                                                                                      ^

C:\Users\Runner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:181:14: error: 'class ESP8266WiFiClass' has no member named 'enableAP'

     if(!WiFi.enableAP(true)) {

              ^

In file included from C:\Users\Runner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:25:0:

C:\Users\Runner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\ESP8266WiFi.h: In member function 'bool ESP8266WiFiAPClass::softAPdisconnect(bool)':

C:\Users\Runner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\ESP8266WiFi.h:390:10: error: 'bool ESP8266WiFiClass::_persistent' is protected

     bool _persistent;

          ^

C:\Users\Runner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:265:13: error: within this context

     if(WiFi._persistent) {

             ^

C:\Users\Runner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:273:57: error: 'DEBUG_WIFI' was not declared in this scope

         DEBUG_WIFI("[APdisconnect] set_config failed!\n");

                                                         ^

C:\Users\Runner\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:277:20: error: 'class ESP8266WiFiClass' has no member named 'enableAP'

         ret = WiFi.enableAP(false);

                    ^

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

tobozo, 2017-11-07:
> the ssid changed but the password does not work

Make sure your password has at least 8 chars

> i still can use the deauther from the oled display but i cant connect to the webpage

You can reset the settings Serial console (no need to flash) by sending the text "reset" from a Serial console while the ESP is connectedt.

wckd, 2017-11-08:
I am getting the same messages trying to compile and flash my D1 Mini.
"Resolved it" by downloading esptool and the version 1.6 .bin-file.

wckd, 2017-11-08:
@statix83 You need version **2.0.0** of ESP8266 to compile this project. 
Your error messages is showing that you have installed version 2.3.0 (latest).

statix83, 2017-11-09:
@tobozo how do i send reset text from a text console ? 
I opejnthe adruino program and write reset ? Thanks

Ah that explains it ... @wckd

tobozo, 2017-11-09:
1) plug the ESP to your USB
2) open Arduino
3) go to tool/ / port and verify the com/ttyUSB port is selected
4) go to tools / Serial console => set speed at 115000
5) type "reset"

statix83, 2017-11-09:
@tobozo thanks for the explanation 

i manage to find the serial monitor ( its the magnifying glass on the right top corner right?

i have followed your instruction and trype reset and click send on the serial monitor it doest do anything 

but i know its conencted .. because there is text keep scrolling ...

statix83, 2017-11-09:
@wckd hey ... thanks 

now i have a new error code while trying to flash the settings.cpp 

Arduino: 1.8.2 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

Archiving built core (caching) in: C:\Users\Runner\AppData\Local\Temp\arduino_cache_883254\core\core_esp8266_esp8266_nodemcuv2_CpuFrequency_80,UploadSpeed_115200,FlashSize_4M3M_823a1f40d4b58642e68ccd9171619c1c.a
Sketch uses 583605 bytes (55%) of program storage space. Maximum is 1044464 bytes.
Global variables use 55300 bytes (67%) of dynamic memory, leaving 26620 bytes for local variables. Maximum is 81920 bytes.
Uploading 587760 bytes from to flash at 0x00000000
.....warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
error: failed sending 0xC0
error: failed sending 8 bytes
error: failed sending 4 bytes
error: failed sending 0xC0
warning: espcomm_send_command: didn't receive command response
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

what did i do wrong this time ?

statix83, 2017-11-09:
not enough memory i guess...

statix83, 2017-11-09:
whoops ... now the board doest want to turn on anymore ...

tobozo, 2017-11-09:
Close the Serial monitor before flashing, sometimes it's interfering.

Also don't forget to select the `Generic ESP8266 Module` board in the boards menu (and choose 1M 256KSPIFFS as flash size) otherwise it'll try to flash using the last selected board (defaults to Arduino Uno) which is a totally different way of flashing and will always fail with espcomm errors.

And not all USB ports can provide enough energy for the ESP to run smoothly, if you're powering from a netbook or a small PC it may not work.

Cables can also be to blame when it suddenly stops working, try with another cable, if using a USB hub, plug the ESP directly to the PC's USB port.

statix83, 2017-11-09:
thanks @tobozo  ... this is part of my learning curve ... hahahaha 
i am buying a new set and will keep in mind with your advice :)

tobozo, 2017-11-09:
Are you sure it's dead?
I've killed an USB port on an old PC just by plugging a regular ESP dev board.
An old version of the DSTIKE build had a heat problem when charging, maybe Travis (the seller on Tindie) can help you troubleshoot that? 
If it happens to be a faulty units he'll send you a brand new one for replacement.

statix83, 2017-11-09:
Its ok ... it was my fault .. the unit was fine when i got it ... but i msssed it up .. i already purchase  a new set 😁

tobozo, 2017-11-10:
I'll close this ticket, feel free to reopen when your purchase is delivered.

closed 2017-11-10 by tobozo