SpacehuhnTech/esp8266_deauther #652: ESP8266 NodeMCU (ESP12E Module) wont work

addisonrudd, 2017-11-10:
Getting this error.

In file included from sketch/Settings.h:12:0,
                 from sketch/APScan.h:9,
                 from sketch/APScan.cpp:1:
/Users/addisonrudd/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0/tools/sdk/include/user_interface.h:456:174: error: declaration of C function 'int wifi_send_pkt_freedom(uint8**, int, bool)' conflicts with
 int wifi_register_send_pkt_freedom_cb(freedom_outside_cb_t cb); void wifi_unregister_send_pkt_freedom__cb(void); int wifi_send_pkt_freedom(uint8 **buf, int len, bool sys_seq);
                                                                                                                                                                              ^
In file included from sketch/Settings.h:12:0,
                 from sketch/APScan.h:9,
                 from sketch/APScan.cpp:1:
/Users/addisonrudd/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0/tools/sdk/include/user_interface.h:453:5: error: previous declaration 'int wifi_send_pkt_freedom(uint8*, int, bool)' here
 int wifi_send_pkt_freedom(uint8 *buf, int len, bool sys_seq);
     ^
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

tobozo, 2017-11-10:
Too much of a hurry to follow all the steps in the [ReadMe](https://github.com/spacehuhn/esp8266_deauther/blob/master/README.md)?

addisonrudd, 2017-11-10:
@tobozo no i read it. i also watched like 5 videos and looked at tutorials. im just new to this so its hard. if you arent going to try to help then dont bother to comment

tobozo, 2017-11-10:
you're getting close, you just skipped a step in the readme, just make sure you follow all of them and it'll work

XAS-712, 2018-01-20:
You may used a library of a wrong version.
Just 2.0 can be used.

tobozo, 2018-03-24:
fixed in the next version, closing this

closed 2018-03-24 by tobozo