SpacehuhnTech/esp8266_deauther #653: Only supplied binaries work on dstike v2.5 OLED board

jgreenpsu, 2017-11-14:
### Hardware/Flashing:
Board: DSTIKE V2.5 OLED
Using OLED:  Tried both with and without.
Using WebServer: YES
Using Serial: YES
Installation mode: SOURCE (works under pre-compiled binary)
Code/binary version: 1.6
Flash size:  Unknown - 1MB?
Flash method: esptool & arduino IDE
Flash Frequency: 160mhz in arduino
Upload Speed: 115200 in arduino
Powered by: USB

### Description:
Describe your problem here, provide as many details as possible.
Make sure you read the FAQ first, incomplete bug reports will be CLOSED without notice.

What you're expecting: Wifi network to show up, If I am compiling with display support, display to work. Serial debug output like normal. The code that I am compiling works great on my nodemcu board.
What you get instead: No response to "reset" over serial, no wifi network and no display output. The board seems to be resetting into some bootloader state.

I have flashed the board with both the arduino IDE and the esptool command line flasher. I have tried all sorts of settings in arduino, of course starting with the settings I could see described in the readme.md.

CMDline for esptool: python ./esptool.py --port=/dev/cu.SLAB_USBtoUART  write_flash  -fm=dio -fs=32m 0x00000 ~/Documents/GitHub/esp8266_deauther/esp8266_deauther/esp8266_deauther.ino.nodemcu.bin

### Debug Messages:

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

jgreenpsu, 2017-11-14:
Ok. I got it to flash source compiled by me by designating a flash size of 1MB on the command line using esptool. I am curious how to get the arduino IDE to play nice, also, I didn't have to designate this on the command line for your binaries to flash and work.

tobozo, 2017-11-14:
In the Arduino IDE I use these settings : 

Board: Generic ESP8266 Module

Flash Size: 1M (256K SPIFFS)

jgreenpsu, 2017-11-15:
Thank you, I can now run code, though this board is still less reliable than my nodemcu board. Here are my settings. Is there something I'm missing? I've been fairly consistently crashing when scanning for accesspoints over the wifi interface.

[image]

tobozo, 2017-11-15:
There's an ongoing but with the same symptoms, long story short: you're probably in a busy area when the scanning returns a lot of access points and triggers the crash.

As a possible workaround you can try to assign a lower value to [maxAPScanResults in ApScan.h[(https://github.com/spacehuhn/esp8266_deauther/blob/master/esp8266_deauther/APScan.h#L4) so the crash will happen less likely.

jgreenpsu, 2017-11-16:
I'm going to open a new issue since this one does not accurately describe the issue I'm having.

closed 2017-11-16 by jgreenpsu

tobozo, 2017-11-16:
Please have a look at the existing issues first :-)