SpacehuhnTech/esp8266_deauther #655: OLED version crash - On scanning for APs via wifi interface

jgreenpsu, 2017-11-16:
### Hardware/Flashing:
Board:DSTIKE & NODEMCU v1.0
Using OLED:  Using USE_DISPLAY: yes
Using WebServer: YES
Using Serial: YES
Installation mode:  both.
Code/binary version: 1.6
Flash size:  1M
Flash method: esptool & arduino IDE
Flash Frequency: in arduino 80mhz
Upload Speed: in arduino 115200
Powered by: Tested on both USB and battery.

### Description:
When using either a release binary or compiled source that uses USE_DISPLAY the board will fault and reset (with stack dump over serial) when scanning for wifi stations from the web interface. Scanning from the display works fine. I have included serial output with scan results and sort logs redacted as the information shouldn't be needed (number of results is still included).

What you're expecting: Scan results (.json) returned successfully.
What you get instead: Serial output shows several scan cycles (without json being sent back) then finally it will stack dump and crash.

How to reproduce: You don't need a display to test this. Just compile with USE_DISPLAY defined and flash. Connect to the deauther wifi and try and scan for accesspoints via the web interface. I have also tested this issue with release binaries which should eliminate any questions regarding my dev environment.

### Debug Messages:
starting AP scan...
MAC - Ch - RSSI - Encrypt. - SSID - Hidden
Scan results: 24
<REDACTED SCAN RESULTS>
scan done
sorting APs 
<REDACTED SORT LOGS>
starting AP scan...
MAC - Ch - RSSI - Encrypt. - SSID - Hidden
Scan results: 24
<REDACTED SCAN RESULTS>
scan done
sorting APs 
<REDACTED SORT LOGS>
starting AP scan...
MAC - Ch - RSSI - Encrypt. - SSID - Hidden
Scan results: 29
<REDACTED SCAN RESULTS>
scan done
sorting APs 
<REDACTED SORT LOGS>
getting attacks JSON (239)...done

Exception (28):
epc1=0x4000bf0e epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff56a0 end: 3fff59a0 offset: 01a0

>>>stack>>>
3fff5840:  3ffe0035 3fff5920 3fff5920 402140a3  
3fff5850:  3fff46a4 3fff3de0 3fff58d0 40214720  
3fff5860:  3fff46a4 3fff3de0 3fff5920 40210d9b  
3fff5870:  00000000 0000004f 0000004f 3fffb8d0  
3fff5880:  0000003f 0000003f 3fffb428 0000001f  
3fff5890:  00000006 00000008 402123f0 3fff4980  
3fff58a0:  3fffa120 0000000f 00000000 3fffa100  
3fff58b0:  0000000f 00000000 3fff7138 0000000f  
3fff58c0:  0000000c 3fff6dd0 0000000f 00000003  
3fff58d0:  3fffb860 0000004f 00000047 4020ec5f  
3fff58e0:  3fff3e1c 3fff46bc 3fff5900 00000001  
3fff58f0:  00000000 00000006 000003e8 4020d888  
3fff5900:  00000000 3fffb2d8 0000000e 3fff4978  
3fff5910:  3fffdc20 000003e9 3fff3de0 4020fd3d  
3fff5920:  3ffe9478 00000000 000003e8 0000f7cb  
3fff5930:  3fff3e00 3fffb2d8 3fff7358 3fff4978  
3fff5940:  3fffdc20 00000000 3fff4971 4020e08e  
3fff5950:  40286dd0 3fff0790 3fff48b4 4020e032  
3fff5960:  00000000 00000000 00000016 40101d0d  
3fff5970:  40212411 00000000 00000000 3fff4978  
3fff5980:  3fffdc20 00000000 3fff4971 40212439  
3fff5990:  00000000 00000000 3fff4980 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

jgreenpsu, 2017-11-16:
After reading other issues, it seems like the webui can cause a crash when out of ram, perhaps the display system eats up some of this ram shorting the time to failure?

This theory still doesn't explain why the scan attempts to run a few times. Perhaps its just the artifact of being completely out of ram.

tobozo, 2017-11-16:
yup, sounds very like heap fragmentation

It can be confirmed by setting HTTP_UPLOAD_BUFLEN value to 1024 in [ESP8266WebServer.h](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/src/ESP8266WebServer.h#L40) and observing that the problem takes longer to show itself.

The more features you disable, the later it happens, but it always ends up happening.

jgreenpsu, 2017-11-16:
Could this be mitigated/eliminated if display/webserver utilities allocated their needed data structures globally instead of at runtime? I'm just brainstorming over here, I'm relatively new to embedded systems.

jgreenpsu, 2017-11-16:
Also, just for clarification this issue makes the webserver on the display enabled board mostly unusable since scanning is broken immediately, though it is still useful for changing settings.

tobozo, 2017-11-16:
Disabling the web server gives the best results.

If you're okay with having only the OLED and buttons as controls, maybe it's worth trying the [serverless](https://github.com/spacehuhn/esp8266_deauther/tree/serverless) branch.

jgreenpsu, 2017-11-17:
I may try that. How open is the development team to potentially merging in my fork if I do a display/menu revamp?

tobozo, 2017-11-17:
The *development team* is @spacehuhn and a v2 is already in progress.
I don't know if it'll be an upgrade or a new project though.

spacehuhn, 2017-11-17:
I will probably keep the current stuff in a different branch and put the new version in the master.
The new version will get a completly new written code base with a better serial, display and web interface support.

tobozo, 2018-03-24:
see if it fixed in the next version and reopen if necessary, closing this

closed 2018-03-24 by tobozo