SpacehuhnTech/esp8266_deauther #660: Wemos D1 Mini Pro binary support

AABatteries, 2017-11-22:
Hi,

I'm having issues trying to upload the binary to Wemos D1 Mini Pro (example product link: [Wemos D1 Mini Pro](https://pl.aliexpress.com/item/WEMOS-D1-mini-Pro-16M-bytes-external-antenna-connector-ESP8266-WIFI-Internet-of-Things-development-board/32724692514.html?ws_ab_test=searchweb0_0,searchweb201602_1_10152_10151_10065_10344_10068_10345_10342_10343_10340_10171_10341_10541_10562_10084_10083_10304_10307_10301_10539_10312_10059_10313_10314_10534_10533_100031_10103_10594_10557_10596_10595_10107_10142,searchweb201603_25,ppcSwitch_5&btsid=440cdfeb-e9e3-4976-87cc-e9c739946205&algo_expid=60fc3f40-3f0e-421d-9bc7-7981c1c3bfa7-2&algo_pvid=60fc3f40-3f0e-421d-9bc7-7981c1c3bfa7&rmStoreLevelAB=2)

This particular board has 16MB of flash memory, and a very handy external antenna U.FL connector. I tried uploading with esptool, using auto settings, settings that worked for every NodeMCU board I tried, but it doesn't work for this board. I manually set the flash size to 16MB, flash freq to 40/80, offset to 0x0 and 0x00000, and while the binary uploads with no issues, unfortunately none of these worked, the board does not run the code. 

Any tips on how to flash this specific board? Thank you for your help and sorry for the unusual formatting!

AABatteries, 2017-11-22:
Ok, I managed to flash the board with esptool using --flash_size 4MB. This fixes the issue I think? Or maybe that additional flash could be used for something?

Jayeshpatil665, 2017-11-23:
With ESP8266 use  Arduno software to flash it...  And use default settings..  Just change your port  ie.  Com 1....100.  Just select right port

spacehuhn, 2018-03-24:
Please try the new version. There is support for the 16mb Wemos pro. That beeing said, it should work with the old version just like that, also when you flash it like a 4mb board.
Upload issues are unrelated to this project, so I will close this now.

closed 2018-03-24 by spacehuhn