SpacehuhnTech/esp8266_deauther #661: Esp8266 isn´t working anymore.

Anonyvius, 2017-11-25:
### Hardware/Flashing:
Board:  Nodemcu
Using OLED:   No
Using WebServer: Yes
Using Serial: No
Installation mode:  binary
Code/binary version: 1.6
Flash size:  1M
Flash method:	 esptool
Flash Frequency:	80Mhz
Upload Speed: 230400
Powered by: USB

### Description:
Describe your problem here, provide as many details as possible.
Make sure you read the FAQ first, incomplete bug reports will be CLOSED without notice.

What you're expecting:  Nodemcu is broken
What you get instead: nothing is working

i flashed the file again and again buts it is not woking anymore. i do not get any WIFI access point. I tried everything. So i am exepting that i broke my esp8266....

tobozo, 2017-11-25:
You need more than bad settings to break your ESP :-)

Please try with Flash Size: 1M(256K SPIFFS)

tobozo, 2018-03-24:
see if it is fixed in the next version, reopen if necessary

closing this

closed 2018-03-24 by tobozo