SpacehuhnTech/esp8266_deauther #667: Bug in attacks.json

louia, 2017-11-29:
### Hardware/Flashing:
Board:				?ESP8266 dev module?Tindie OLED?Wemos Mini?Other (specify)?
Using OLED:   NO
Using WebServer: YES
Using Serial: NO
Installation mode:  source
Code/binary version: 1.6
Flash size:  ?1M
Flash method:	Arduino IDE
Flash Frequency:	
Upload Speed: 115200
Powered by:  Battery

### Description:
When i start an attack by deauth, sometime i must reboot my card because a message say on my webphone "Attack.json failed".

What you're expecting:  ---
What you get instead: ---

### Debug Messages: Attack.json failed
```
Put the serial output here

marek26340, 2017-12-06:
Have you tried another browser?

louia, 2017-12-06:
No, just chrome on android

marek26340, 2017-12-26:
Try to find the channel of the network that you are trying to attack, set the channel of your deauther to the channel of the AP that you are trying to deauth, and try to attack it again. Did you get that error again?

cambgiles, 2017-12-28:
I have been having this issue as well.

bert4, 2018-01-05:
Same here, except I am also getting it on ClientScan.json

RudyTheKid, 2018-01-12:
I get an error loading attackinfo.json

tobozo, 2018-03-24:
see if it is fixed in the next version, reopen if necessary

closing this

closed 2018-03-24 by tobozo

spacehuhn, 2018-03-24:
You can loose connection when starting an attack, as explained in the web interface.