SpacehuhnTech/esp8266_deauther #668: Error compiling for board NodeMCU 1.0 (ESP-12E Module)

OGBRTC, 2017-12-01:
### Hardware/Flashing:
Board:				?ESP8266 dev module?Tindie OLED?Wemos Mini?Other (specify)?
Using OLED:   NO
Using WebServer: NO
Using Serial: YES
Installation mode:  source
Code/binary version: ?1.5?1.6?
Flash size:  4M
Flash method:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						115200
Powered by:  USB

### Description:
Trying to verify it before uploading and it said it had an error compiling the board. This is the error message.
Arduino: 1.8.5 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Serial, 115200, 4M (3M SPIFFS)"

C:\Users\billy\AppData\Local\Temp\arduino_modified_sketch_907039\esp8266_deauther.ino:66:18: fatal error: data.h: No such file or directory

 #include "data.h"

                  ^

compilation terminated.

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

What you're expecting:  To be able to verify and upload
What you get instead: Not able to verify it
### Debug Messages:
```
COM 3

marek26340, 2017-12-06:
That is weird. It is unable to find the "data.h" library. I don't know how to fix that, because I haven't used Arduino IDE in a loooong time. Things you could try: Try running Arduino IDE as administrator.; Also make sure that the library "data.h" is in the libraries folder of Arduino IDE. If it isn't there, or you don't have it, then you should be able to find it on the internet.

I am really just a beginner, this help might not work. People with more knowledge around this should be able to help you better.

tobozo, 2017-12-06:
@marek26340 I guess @OGBRTC only used the esp8266_deauther.ino and created a new project instead of opening it from the file menu.

Most basic examples available in the Arduino IDE fit in a single file so it's easy to be misled at first.

Closing this issue as it is not an esp8266_deauther problem.

closed 2017-12-06 by tobozo

anamikasingh313, 2018-05-23:
oard
we are getting the compilation error by including some headerfiles in node mcu b

Rofisetiawan17, 2018-12-13:
Arduino: 1.8.8 (Windows Store 1.8.19.0) (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Enabled, 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

c:/users/rofi setiawan/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\Rofi Setiawan\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta1/tools/sdk/lib\libmain.a(app_main.o): in function `flash_data_check':

(.irom0.text+0x7f1): undefined reference to `pvPortZalloc'

c:/users/rofi setiawan/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: (.irom0.text+0x86e): undefined reference to `pvPortZalloc'

c:/users/rofi setiawan/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: (.irom0.text+0x9ff): undefined reference to `vPortFree'

c:/users/rofi setiawan/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: (.irom0.text+0xaae): undefined reference to `vPortFree'

c:/users/rofi setiawan/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\Rofi Setiawan\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta1/tools/sdk/lib\libmain.a(eagle_lwip_if.o):(.irom0.text+0xdf): undefined reference to `pvPortMalloc'

c:/users/rofi setiawan/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\Rofi Setiawan\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta1/tools/sdk/lib\libmain.a(eagle_lwip_if.o): in function `eagle_lwip_if_alloc':

(.irom0.text+0x184): undefined reference to `pvPortMalloc'

c:/users/rofi setiawan/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: (.irom0.text+0x24b): undefined reference to `pvPortMalloc'

c:/users/rofi setiawan/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: (.irom0.text+0x2a8): undefined reference to `vPortFree'

c:/users/rofi setiawan/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: (.irom0.text+0x2db): undefined reference to `vPortFree'

c:/users/rofi setiawan/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: (.irom0.text+0x2f2): undefined reference to `vPortFree'

c:/users/rofi setiawan/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\Rofi Setiawan\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta1/tools/sdk/lib\libmain.a(user_interface.o):(.text+0x14): undefined reference to `xPortWantedSizeAlign'

c:/users/rofi setiawan/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\Rofi Setiawan\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta1/tools/sdk/lib\libmain.a(user_interface.o):(.text+0x20): undefined reference to `pvPortMalloc'

c:/users/rofi setiawan/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\Rofi Setiawan\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta1/tools/sdk/lib\libmain.a(user_interface.o):(.text+0x24): undefined reference to `vPortFree'

c:/users/rofi setiawan/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\Rofi Setiawan\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta1/tools/sdk/lib\libmain.a(user_interface.o): in function `os_printf_plus':

(.text+0x94): undefined reference to `xPortWantedSizeAlign'

c:/users/rofi setiawan/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: (.text+0xf5): undefined reference to `pvPortMalloc'

c:/users/rofi setiawan/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: (.text+0x146): undefined reference to `vPortFree'

c:/users/rofi setiawan/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\Rofi Setiawan\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta1/tools/sdk/lib\libmain.a(user_interface.o): in function `system_restart_local':

[... cut ...]
c:/users/rofi setiawan/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\Rofi Setiawan\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta1/tools/sdk/lib\libpp.a(esf_buf.o):(.irom0.text+0x2e9): more undefined references to `vPortFree' follow

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

.
.
.
How to solve this?

forseenink3238, 2018-12-13:
Sorry can't find any way except flashing it with the bin file.

On 3:52pm, Thu, 13 Dec 2018 Rofisetiawan17 <notifications@github.com wrote:

> Arduino: 1.8.8 (Windows Store 1.8.19.0) (Windows 10), Board: "NodeMCU 1.0
> (ESP-12E Module), 80 MHz, Flash, Enabled, 4M (no SPIFFS), v2 Lower Memory,
> Disabled, None, Only Sketch, 115200"
>
> c:/users/rofi
> setiawan/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe:
> C:\Users\Rofi
> Setiawan\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta1/tools/sdk/lib\libmain.a(app_main.o):
> in function `flash_data_check':
>
> (.irom0.text+0x7f1): undefined reference to `pvPortZalloc'
>
> c:/users/rofi
> setiawan/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe:
> (.irom0.text+0x86e): undefined reference to `pvPortZalloc'
>
> c:/users/rofi
> setiawan/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe:
> (.irom0.text+0x9ff): undefined reference to `vPortFree'
>
> c:/users/rofi
> setiawan/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe:
> (.irom0.text+0xaae): undefined reference to `vPortFree'
>
> c:/users/rofi
> setiawan/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe:
> C:\Users\Rofi
> Setiawan\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta1/tools/sdk/lib\libmain.a(eagle_lwip_if.o):(.irom0.text+0xdf):
> undefined reference to `pvPortMalloc'
>
> c:/users/rofi
> setiawan/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe:
> C:\Users\Rofi
> Setiawan\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta1/tools/sdk/lib\libmain.a(eagle_lwip_if.o):
> in function `eagle_lwip_if_alloc':
>
> (.irom0.text+0x184): undefined reference to `pvPortMalloc'
>
> c:/users/rofi
> setiawan/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe:
> (.irom0.text+0x24b): undefined reference to `pvPortMalloc'
>
> c:/users/rofi
> setiawan/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe:
> (.irom0.text+0x2a8): undefined reference to `vPortFree'
>
> c:/users/rofi
> setiawan/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe:
> (.irom0.text+0x2db): undefined reference to `vPortFree'
>
> c:/users/rofi
> setiawan/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe:
> (.irom0.text+0x2f2): undefined reference to `vPortFree'
>
> c:/users/rofi
> setiawan/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe:
> C:\Users\Rofi
> Setiawan\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta1/tools/sdk/lib\libmain.a(user_interface.o):(.text+0x14):
> undefined reference to `xPortWantedSizeAlign'
>
> c:/users/rofi
> setiawan/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe:
> C:\Users\Rofi
> Setiawan\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta1/tools/sdk/lib\libmain.a(user_interface.o):(.text+0x20):
> undefined reference to `pvPortMalloc'
>
[... 989 lines cut ...]
> undefined reference to `uart_get_debug'
>
> c:/users/rofi
> setiawan/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe:
> C:\Users\ROFISE~1\AppData\Local\Temp\arduino_cache_780227\core\core_ae1f5361da39f830aea0f23520a55dfe.a(HardwareSerial.cpp.o):(.text._ZN14HardwareSerial3endEv+0x4):
> undefined reference to `uart_set_debug'
>
> c:/users/rofi
> setiawan/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe:
> C:\Users\ROFISE~1\AppData\Local\Temp\arduino_cache_780227\core\core_ae1f5361da39f830aea0f23520a55dfe.a(HardwareSerial.cpp.o):(.text._ZN14HardwareSerial3endEv+0x8):
> undefined reference to `uart_uninit'
>
> c:/users/rofi
> setiawan/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe:
> C:\Users\ROFISE~1\AppData\Local\Temp\arduino_cache_780227\core\core_ae1f5361da39f830aea0f23520a55dfe.a(HardwareSerial.cpp.o):
> in function `HardwareSerial::end()':
>
> C:\Users\Rofi
> Setiawan\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta1\cores\esp8266/HardwareSerial.cpp:54:
> undefined reference to `uart_get_debug'
>
> c:/users/rofi
> setiawan/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-2-59d892c8/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe:
> C:\Users\Rofi
> Setiawan\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.5.0-beta1\cores\es

karanjeet1, 2019-01-17:
this error basically arise when the libraries are not compatible with the code.
mainly when you update for library from previous versions to new version then this error comes.
so its suggested to edit your code according to updated library or downgrade you library.

Thank you

rohantehalyani, 2019-02-03:
C:\Users\Rohan Tehalyani\Documents\Arduino\libraries\Blynk\src/BlynkSimpleEsp8266.h:18:21: fatal error: version.h: No such file or directory

 #include <version.h>

                     ^

compilation terminated.

Using library ESP8266WiFi at version 1.0 in folder: C:\Users\Rohan Tehalyani\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0-rc2\libraries\ESP8266WiFi 
**Using library Blynk at version 0.6.0 in folder: C:\Users\Rohan Tehalyani\Documents\Arduino\libraries\Blynk 
exit status 1**
**Error compiling for board NodeMCU 0.9 (ESP-12 Module).**

swapnavalluri, 2019-02-13:
Arduino: 1.8.3 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Enabled, 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

In file included from C:\Users\DELL\Documents\Arduino\libraries\firebase-arduino-master\src/Firebase.h:30:0,

                 from C:\Users\DELL\Documents\Arduino\libraries\firebase-arduino-master\src/FirebaseArduino.h:22,

                 from C:\Users\DELL\Documents\Arduino\FirebaseDemo_ESP8266\FirebaseDemo_ESP8266.ino:22:

C:\Users\DELL\Documents\Arduino\libraries\firebase-arduino-master\src/FirebaseObject.h:109:21: error: 'StaticJsonBuffer' was not declared in this scope

     std::shared_ptr<StaticJsonBuffer<FIREBASE_JSONBUFFER_SIZE>> buffer_;

                     ^

In file included from C:\Users\DELL\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:10:0,

                 from C:\Users\DELL\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from C:\Users\DELL\Documents\Arduino\libraries\firebase-arduino-master\src/Firebase.h:26,

                 from C:\Users\DELL\Documents\Arduino\libraries\firebase-arduino-master\src/FirebaseArduino.h:22,

                 from C:\Users\DELL\Documents\Arduino\FirebaseDemo_ESP8266\FirebaseDemo_ESP8266.ino:22:

C:\Users\DELL\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Object/ObjectRef.hpp:13:69: error: template argument 1 is invalid

   ((NUMBER_OF_ELEMENTS) * sizeof(ARDUINOJSON_NAMESPACE::VariantSlot))

                                                                     ^

C:\Users\DELL\Documents\Arduino\libraries\firebase-arduino-master\src/FirebaseObject.h:28:34: note: in expansion of macro 'JSON_OBJECT_SIZE'

 #define FIREBASE_JSONBUFFER_SIZE JSON_OBJECT_SIZE(32)

                                  ^

C:\Users\DELL\Documents\Arduino\libraries\firebase-arduino-master\src/FirebaseObject.h:109:38: note: in expansion of macro 'FIREBASE_JSONBUFFER_SIZE'

     std::shared_ptr<StaticJsonBuffer<FIREBASE_JSONBUFFER_SIZE>> buffer_;

                                      ^

In file included from C:\Users\DELL\Documents\Arduino\libraries\firebase-arduino-master\src/Firebase.h:30:0,

                 from C:\Users\DELL\Documents\Arduino\libraries\firebase-arduino-master\src/FirebaseArduino.h:22,

                 from C:\Users\DELL\Documents\Arduino\FirebaseDemo_ESP8266\FirebaseDemo_ESP8266.ino:22:

C:\Users\DELL\Documents\Arduino\libraries\firebase-arduino-master\src/FirebaseObject.h:109:62: error: expected unqualified-id before '>' token

     std::shared_ptr<StaticJsonBuffer<FIREBASE_JSONBUFFER_SIZE>> buffer_;

                                                              ^

In file included from C:\Users\DELL\Documents\Arduino\libraries\firebase-arduino-master\src/FirebaseArduino.h:22:0,

                 from C:\Users\DELL\Documents\Arduino\FirebaseDemo_ESP8266\FirebaseDemo_ESP8266.ino:22:

C:\Users\DELL\Documents\Arduino\libraries\firebase-arduino-master\src/Firebase.h:86:19: error: 'StaticJsonBuffer' was not declared in this scope

   std::shared_ptr<StaticJsonBuffer<FIREBASE_JSONBUFFER_SIZE>> buffer_;

                   ^

In file included from C:\Users\DELL\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:10:0,

                 from C:\Users\DELL\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from C:\Users\DELL\Documents\Arduino\libraries\firebase-arduino-master\src/Firebase.h:26,

                 from C:\Users\DELL\Documents\Arduino\libraries\firebase-arduino-master\src/FirebaseArduino.h:22,

                 from C:\Users\DELL\Documents\Arduino\FirebaseDemo_ESP8266\FirebaseDemo_ESP8266.ino:22:

C:\Users\DELL\Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Object/ObjectRef.hpp:13:69: error: template argument 1 is invalid

   ((NUMBER_OF_ELEMENTS) * sizeof(ARDUINOJSON_NAMESPACE::VariantSlot))

                                                                     ^

C:\Users\DELL\Documents\Arduino\libraries\firebase-arduino-master\src/FirebaseObject.h:28:34: note: in expansion of macro 'JSON_OBJECT_SIZE'

 #define FIREBASE_JSONBUFFER_SIZE JSON_OBJECT_SIZE(32)

                                  ^

C:\Users\DELL\Documents\Arduino\libraries\firebase-arduino-master\src/Firebase.h:86:36: note: in expansion of macro 'FIREBASE_JSONBUFFER_SIZE'

   std::shared_ptr<StaticJsonBuffer<FIREBASE_JSONBUFFER_SIZE>> buffer_;

                                    ^

In file included from C:\Users\DELL\Documents\Arduino\libraries\firebase-arduino-master\src/FirebaseArduino.h:22:0,

                 from C:\Users\DELL\Documents\Arduino\FirebaseDemo_ESP8266\FirebaseDemo_ESP8266.ino:22:

C:\Users\DELL\Documents\Arduino\libraries\firebase-arduino-master\src/Firebase.h:86:60: error: expected unqualified-id before '>' token

   std::shared_ptr<StaticJsonBuffer<FIREBASE_JSONBUFFER_SIZE>> buffer_;

                                                            ^

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
     I GOT THIS ERROR WHILE COMPILING THE  FIREBASE DEMO CODE ON ARDUINO IDE

bisalhao, 2019-03-11:
Dá-me sempre este erro, já não sei o que fazer.

********************************************************************************************************
Arduino: 1.8.8 (Windows 7), Placa:"NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled, 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

In file included from C:\Users\Luis\Desktop\WEATHER\WEATHER.ino:7:0:

WundergroundForecast.h:28:26: error: JsonListener.h: No such file or directory

 #include <JsonListener.h>

                          ^

compilation terminated.

exit status 1
JsonListener.h: No such file or directory
*******************************************************************************************************

GamesOri, 2019-03-21:
Arduino: 1.8.5 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"

Build options changed, rebuilding all
In file included from C:\Users\GamesOri\Documents\Arduino\libraries\firebase-arduino-master\src/Firebase.h:30:0,

                 from C:\Users\GamesOri\Documents\Arduino\libraries\firebase-arduino-master\src/FirebaseArduino.h:22,

                 from E:\Arduino Code\NodeContFirebase\NodeContFirebase.ino:6:

C:\Users\GamesOri\Documents\Arduino\libraries\firebase-arduino-master\src/FirebaseObject.h:109:11: error: StaticJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6

     std::shared_ptr<StaticJsonBuffer<FIREBASE_JSONBUFFER_SIZE>> buffer_;

           ^

In file included from C:\Users\GamesOri\Documents\Arduino\libraries\firebase-arduino-master\src/FirebaseArduino.h:22:0,

                 from E:\Arduino Code\NodeContFirebase\NodeContFirebase.ino:6:

C:\Users\GamesOri\Documents\Arduino\libraries\firebase-arduino-master\src/Firebase.h:86:11: error: StaticJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6

   std::shared_ptr<StaticJsonBuffer<FIREBASE_JSONBUFFER_SIZE>> buffer_;

           ^

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

digra33, 2019-05-31:
if the trouble is ''esptool.fatalerror:Failed to connect to ESP8266 :Timed out waiting for packet header'',. what must i do ?

digra33, 2019-05-31:
Help me pleae

VivekSharma184, 2019-09-28:
bhut hard..

ldq1709, 2020-09-27:
Can somebody help how to fix this erro please!!! And i am not so good at grammar, please forgive me about this.

Arduino: 1.8.14 Hourly Build 2020/06/24 01:33 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled, Basic SSL ciphers (lower ROM use), 4M (2M SPIFFS), v2 Lower Memory, Disabled, None, Sketch + WiFi Settings, 115200"

In file included from c:\programdata\matlab\supportpackages\r2017b\3p.instrset\arduinoide.instrset\idepkgs\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\algorithm:60:0,

             from C:\ProgramData\MATLAB\SupportPackages\R2017b\3P.instrset\arduinoide.instrset\idepkgs\packages\esp8266\hardware\esp8266\2.5.2\cores\esp8266/Arduino.h:257,

             from sketch\esp.ino.cpp:1:
c:\programdata\matlab\supportpackages\r2017b\3p.instrset\arduinoide.instrset\idepkgs\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9\xtensa-lx106-elf\include\c++\4.8.2\utility:68:28: fatal error: bits/c++config.h: No such file or directory

#include <bits/c++config.h>

                        ^
compilation terminated.

exit status 1

Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Bommlon, 2020-10-09:
@ldq1709 please open a new issue for your problem to make it easier to find by others who have the same problem.

SHERAMII, 2021-04-18:
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
 why this error

SHERAMII, 2021-04-18:
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

I am getting this error, I am using adafruit.io and ESP8266. can I know any solutions. I am beginner so any advice would be useful.

HiBijoy143, 2021-11-01:
Arduino: 1.8.16 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled (new aborts on oom), Disabled, All SSL ciphers (most compatible), 32KB cache + 32KB IRAM (balanced), Use pgm_read macros for IRAM/PROGMEM, 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Hasan Imam Bijoy\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Hasan Imam Bijoy\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries E:\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:xtal=80,vt=flash,exception=disabled,stacksmash=disabled,ssl=all,mmu=3232,non32xfer=fast,eesz=4M2M,led=2,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -vid-pid=10C4_EA60 -ide-version=10816 -build-path C:\Users\HASANI~1\AppData\Local\Temp\arduino_build_655499 -warnings=none -build-cache C:\Users\HASANI~1\AppData\Local\Temp\arduino_cache_849404 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Hasan Imam Bijoy\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed -prefs=runtime.tools.xtensa-lx106-elf-gcc-3.0.4-gcc10.3-1757bed.path=C:\Users\Hasan Imam Bijoy\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed -prefs=runtime.tools.mkspiffs.path=C:\Users\Hasan Imam Bijoy\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\3.0.4-gcc10.3-1757bed -prefs=runtime.tools.mkspiffs-3.0.4-gcc10.3-1757bed.path=C:\Users\Hasan Imam Bijoy\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\3.0.4-gcc10.3-1757bed -prefs=runtime.tools.mklittlefs.path=C:\Users\Hasan Imam Bijoy\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\3.0.4-gcc10.3-1757bed -prefs=runtime.tools.mklittlefs-3.0.4-gcc10.3-1757bed.path=C:\Users\Hasan Imam Bijoy\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\3.0.4-gcc10.3-1757bed -prefs=runtime.tools.python3.path=C:\Users\Hasan Imam Bijoy\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\Hasan Imam Bijoy\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -verbose E:\Arduino\sketch_nov02agass\sketch_nov02agass.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Hasan Imam Bijoy\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Hasan Imam Bijoy\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries E:\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:xtal=80,vt=flash,exception=disabled,stacksmash=disabled,ssl=all,mmu=3232,non32xfer=fast,eesz=4M2M,led=2,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=115200 -vid-pid=10C4_EA60 -ide-version=10816 -build-path C:\Users\HASANI~1\AppData\Local\Temp\arduino_build_655499 -warnings=none -build-cache C:\Users\HASANI~1\AppData\Local\Temp\arduino_cache_849404 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Hasan Imam Bijoy\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed -prefs=runtime.tools.xtensa-lx106-elf-gcc-3.0.4-gcc10.3-1757bed.path=C:\Users\Hasan Imam Bijoy\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\3.0.4-gcc10.3-1757bed -prefs=runtime.tools.mkspiffs.path=C:\Users\Hasan Imam Bijoy\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\3.0.4-gcc10.3-1757bed -prefs=runtime.tools.mkspiffs-3.0.4-gcc10.3-1757bed.path=C:\Users\Hasan Imam Bijoy\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\3.0.4-gcc10.3-1757bed -prefs=runtime.tools.mklittlefs.path=C:\Users\Hasan Imam Bijoy\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\3.0.4-gcc10.3-1757bed -prefs=runtime.tools.mklittlefs-3.0.4-gcc10.3-1757bed.path=C:\Users\Hasan Imam Bijoy\AppData\Local\Arduino15\packages\esp8266\tools\mklittlefs\3.0.4-gcc10.3-1757bed -prefs=runtime.tools.python3.path=C:\Users\Hasan Imam Bijoy\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -prefs=runtime.tools.python3-3.7.2-post1.path=C:\Users\Hasan Imam Bijoy\AppData\Local\Arduino15\packages\esp8266\tools\python3\3.7.2-post1 -verbose E:\Arduino\sketch_nov02agass\sketch_nov02agass.ino

Using board 'nodemcuv2' from platform in folder: C:\Users\Hasan Imam Bijoy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2

Using core 'esp8266' from platform in folder: C:\Users\Hasan Imam Bijoy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2

Detecting libraries used...

"C:\\Users\\Hasan Imam Bijoy\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\3.0.4-gcc10.3-1757bed/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -D_GNU_SOURCE "-IC:\\Users\\Hasan Imam Bijoy\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.2/tools/sdk/include" "-IC:\\Users\\Hasan Imam Bijoy\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\3.0.2/tools/sdk/lwip2/include" "-IC:\\Users\\Hasan Imam Bijoy\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware [... cut ...]
Using library ESP8266WiFi at version 1.0 in folder: C:\Users\Hasan Imam Bijoy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266WiFi 

Using library firebase-arduino-master at version 0.1 in folder: E:\Arduino\libraries\firebase-arduino-master 

Using library ESP8266HTTPClient at version 1.2 in folder: C:\Users\Hasan Imam Bijoy\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\libraries\ESP8266HTTPClient 

exit status 1

Error compiling for board NodeMCU 1.0 (ESP-12E Module).

Sardashti, 2021-11-08:
Hi, you must on "Board Manager" , remove and again install the last version of esp8266 .
good lucke

ALifHaider19, 2022-02-08:
Arduino: 1.8.19 (Windows Store 1.8.57.0) (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled (new aborts on oom), Disabled, All SSL ciphers (most compatible), 32KB cache + 32KB IRAM (balanced), Use pgm_read macros for IRAM/PROGMEM, 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"

This is my error message!
I tried to fix this problem, but I failed. I would be glad if you can help me
C:\Users\u1909\OneDrive\Documents\Arduino\IOT_tutorial_JSON_read\IOT_tutorial_JSON_read.ino: In function 'void loop()':

IOT_tutorial_JSON_read:28:15: error: call to 'HTTPClient::begin' declared with attribute error: obsolete API, use ::begin(WiFiClient, url)

   28 |     http.begin("http://jsonplaceholder.typicode.com/users/1");

      |     ~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

exit status 1

call to 'HTTPClient::begin' declared with attribute error: obsolete API, use ::begin(WiFiClient, url)

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

adedotunowojori, 2022-12-26:
I had some of the earlier issues when compiling my nodeMCU code. The way i went round it was deactivating my antivirus during the period of compilation.