SpacehuhnTech/esp8266_deauther #672: pwned network  is  not showing up

noahnow2004, 2017-12-06:
I was able to do this a month ago when I got this thing, but then a week ago when put a check on the setting, 'Hide SSID' I have never been able to find the network it creates, 'pwned'. I deleted the program and then put it back on but it still does not work. PLEASE HELP. I NEED IT FAST

PxLtommy, 2017-12-08:
When in network settings on another device, you need to click 'other', and enter its name, security etc

MrAppAndCrap, 2017-12-10:
Connect the board to your pc via usb. Flash the .bin files (https://github.com/spacehuhn/esp8266_deauther#installation) After you finish flashing unplug the device, plug it into a phone charger socket or a 3.0+ usb port on your computer. You should see the wifi network now. The reason for plugging it into a good phone charger is because it might not be pulling enough power from usb ports on pc's.

Johnsjs232, 2018-01-05:
I am having the same problem I flashed my esp8266 with nodemcu firmware programmer it said it was successful but when I went to use it it would not give off a wifi signal. I have no idea what to do to fix it

Johnsjs232, 2018-01-05:
I tried plugging it into the wall too

tobozo, 2018-03-24:
see if it is fixed in the next version, reopen if necessary

closing this

closed 2018-03-24 by tobozo

fankaisong, 2018-05-05:
I have the same issue. I just bought it today and followed all instruction. The pwned just not showing after it.  I tried "Blinking" example on Arduino and it worked.  So, the other thing was that when I runed EEPROM earase from Arduino, the pwned wifi showed up

rogaha, 2018-05-12:
same issue here