SpacehuhnTech/esp8266_deauther #675: MAC ADDRESS ASSIGNMENT ERROR

khwilo, 2017-12-13:
I noticed that the mac address for one phone that I was performing a de-authentication attack on was not consistent with that of the manufacturer. Could that be an issue of a counterfeit phone (which I highly doubt)? Or could it be an error in the program? The mobile phone in question is INFINIX NOTE 4 which is assigned TecnoMob... as it's station's MAC address.

spacehuhn, 2017-12-13:
What's the first 3 bytes of the mac address?

khwilo, 2017-12-21:
The first three bytes of the MAC address is 78:ff:ca which is of Tecno mobile. It seems I may have to check on the device specifications then. What do you reckon?

tobozo, 2018-03-24:
see if it is fixed in the next version, reopen if necessary

closing this

closed 2018-03-24 by tobozo