SpacehuhnTech/esp8266_deauther #680: The module itself creates a network wifi

konstantin42127, 2017-12-30:
I don't understand how this happens ....
but nonetheless
After your firmware it worked fine, after about a month I decided to return the old firmware (via ESP8266Flasher.exe). But after that strange things began to happen
I programmed this module through arduino IDle for my project, but after a while noticed that my esp8266 started to create Wi-Fi networks!
In addition, my esp8266 does only one wifi network and it completely replicates my home network with the same MAC address, password and ssid
How to fix it? and in general, how is this possible?

marek26340, 2018-01-05:
Wow how could this be possible??? I don't understand. But you can try to erase the entire flash memory of the ESP8266. Download 
[blank_bin_esp8266.zip](https://github.com/spacehuhn/esp8266_deauther/files/1605833/blank_bin_esp8266.zip)
unzip it and open the ESP8266flasher, and flash the blank.bin to 0x000000, 0x100000, 0x200000 and 0x300000. Now your ESP's flash memory will be erased, so now you need to reupload the deauther firmware. After that tell us if it is behaving normally now.

konstantin42127, 2018-01-05:
@marek26340 , I did as you said, I flash the blank.bin to 0x000000, 0x100000 (0x200000 and 0x300000 was not listed). But the problem remained

marek26340, 2018-01-05:
You can type in 0x300000 manually.

konstantin42127, 2018-01-05:
@marek26340 , I did everything, nothing has changed

tobozo, 2018-01-05:
Don't enter your personal wifi SSID/Password in the esp8266_deauther.ino code, you're supposed to leave those settings as they are, or use a unique SSID and a password with at least 8 chars.

konstantin42127, 2018-01-06:
I have formatted esp8266 several times and now it has ceased to create my home network. Instead, it creates a "pwned" network... _I again do not understand anything_

tobozo, 2018-01-06:
@konstantin42127 it's a good thing you admit you don't understand something.
Actually it's the best step you can take because it helps you identifying where you should document yourself (i.e by reading the FAQ or by studying WiFi in general).

marek26340, 2018-01-06:
@tobozo btw he was flashing it via ESP8266flasher, so he used binary files, he was not building from source.

@konstantin42127 So this means now that the deauther started working normally? 
Connect to the WiFi network "pwned" with password "deauther", open your browser and type in "192.168.4.1". That should bring up the Web interface, but you might already know what to do now. 
Is it working now? (deauth, beacon flood etc.)

konstantin42127, 2018-01-06:
In general, when I only formatted Esp8266, there were no networks. But after I uploaded the sketch through arduino, the module began to create a network "pwned".
After, if I go to "192.168.4.1" then I see the site that I uploaded in the sketch via arudino

Here's a screenshot:
[image]

Here's the sketch I'm uploading:

```
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>

const char *ssid = "Lena"; 
const char *password = "al555"; 

ESP8266WebServer server(80);   //Web server object. Will be listening in port 80 (default for HTTP)
class Step
{
  public:
    int time;
    int temp;
};

class Mode
{
  public:
    String name;
    int stepsCount;
    Step steps[10];
};

Mode modes[10];
int modesCount = 0;
String ipLocal = "192.168.0.106";
String html = "<!DOCTYPE HTML> <html> <head> <meta name=\"viewport\" content=\"width=device-width, initial-scale=1\"> <p><input name=\"type\" type=\"radio\" value=\"new\" onclick=\"typeClick(this);\"> Create new <input name=\"type\" type=\"radio\" value=\"old\" onclick=\"typeClick(this);\"> Select existing </p> <p><input id = \"number\" type=\"text\" name=\"number\" placeholder=\"Number of the mode\"></p> <p><input id = \"name\" type=\"text\" name=\"name\" placeholder=\"Mode name\"></p> <p><input id=\"buttonAdd\" type=\"button\" value=\"Add\" onclick=\"creater()\"></p> <p><input type=\"button\" value=\"Send\" onclick=\"sendGet()\"></p> </head> <body> <script type=\"text/javascript\"> var type = \"old\"; var name = \"mode_default\"; var number = 0; document.getElementById(\"buttonAdd\").style.display = \"none\"; document.getElementById(\"name\").style.display = \"none\"; document.getElementById(\"number\").style.display = \"none\"; var inputName = document.getElementById(\"name\"); inputName.oninput = function() { name = inputName.value; }; var inputNumber = document.getElementById(\"number\"); inputNumber.oninput = function() { number = inputNumber.value; }; var count = 0; function creater(){ count++; var d=document.createElement('div'); d.id='div'+count.toString(); var strdivid = \"'\"+'div'+count.toString()+\"'\"; d.innerHTML = '<input id=\"time'+count.toString()+'\" type=\"text\" name=\"time'+count.toString()+'\"> <input id=\"temp'+count.toString()+'\" type=\"text\" name=\"temp'+count.toString()+'\"> <input type=\"button\" value=\"x\" onclick=\"deleater('+strdivid+')\"><br/><br/>'; document.body.appendChild(d); } function deleater(idel){ var element = document.getElementById(idel); while (element.firstChild) { element.removeChild(element.firstChild); } element.remove(); } function sendGet(){ var xhr = new XMLHttpRequest(); var params = \"\"; if(type==\"new\"){ params = 'type=' + encodeURIComponent(type) + '&name=' + encodeURIComponent(name); for (var i = 1; i < count+1; i++) { if(document.getElementById(\"time\"+i.toString())!=null){ params+='&step'+i.toString()+'='+document.getElementById(\"time\"+i.toString()).value+'x'+document.getElementById(\"temp\"+i.toString()).value+'x'; } } } else{ params = 'type=' + encodeURIComponent(type) + '&number=' + encodeURIComponent(number); } xhr.open(\"GET\", 'http://" + ipLocal + "/genericArgs?' + params, true); xhr.send(); } function typeClick(myRadio) { if(myRadio.value==\"new\"){ document.getElementById(\"buttonAdd\").style.display = \"block\"; document.getElementById(\"name\").style.display = \"block\"; document.getElementById(\"number\").style.display = \"none\"; type=\"new\"; } else{ document.getElementById(\"buttonAdd\").style.display = \"none\"; document.getElementById(\"name\").style.display = \"none\"; document.getElementById(\"number\").style.display = \"block\"; type=\"old\"; } } </script> </body> </html>";

void handleRoot() {
  server.send(200, "text/html", html);
}

void handleGenericArgs() { //Handler http://192.168.0.106/genericArgs?type=new&name=mode0&part1=60x100x

  //type=new&name=someName&60=100 **GET

  String message = "";

  if (server.arg(0) == "new") {
    for (int j = 0; j < 9; j++) {
      if (modes[j].name != server.arg(1)) {
        Mode m;
        m.name = server.arg(1);
        Serial.println("\n\nСоздан новый режим с иминем " + m.name);
        Step s[10];
        m.stepsCount = server.args() - 2;
        for (int i = 2; i < server.args(); i++) {
          Step tempS;
          Serial.println("Период №" + (String)(i - 1));

          String tempStr = "";
          int arrN = 0;
          String stringSteps = server.arg(i);
          char c = 'x';
          for (int k = 0; k < stringSteps.length(); k++) {
            if (stringSteps[k] == c) arrN++;
          }
          String returns[arrN];
          int returnsCount = 0;
          for (int k = 0; k < stringSteps.length(); k++) {
            if (stringSteps[k] == c) {
              returns[returnsCount] = tempStr;
[... 45 lines cut ...]
  }

  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());  //Print the local IP to access the server
  ipLocal = WiFi.localIP().toString();
  Serial.println(WiFi.localIP().toString());
  
  html = "<!DOCTYPE HTML> <html> <head> <meta name=\"viewport\" content=\"width=device-width, initial-scale=1\"> <p><input name=\"type\" type=\"radio\" value=\"new\" onclick=\"typeClick(this);\"> Create new <input name=\"type\" type=\"radio\" value=\"old\" onclick=\"typeClick(this);\"> Select existing </p> <p><input id = \"number\" type=\"text\" name=\"number\" placeholder=\"Number of the mode\"></p> <p><input id = \"name\" type=\"text\" name=\"name\" placeholder=\"Mode name\"></p> <p><input id=\"buttonAdd [... cut ...] r\").style.display = \"block\"; type=\"old\"; } } </script> </body> </html>";

  server.on("/", handleRoot);
  server.on("/genericArgs", handleGenericArgs); //Associate the handler function to the path
  //server.on("/specificArgs", handleSpecificArg);   //Associate the handler function to the path

  server.begin();                                       //Start the server
  Serial.println("Server listening");

}

void loop() {

  server.handleClient();    //Handling of incoming requests

}

```

tobozo, 2018-03-24:
see if it is fixed in the next version, reopen if necessary

closing this

closed 2018-03-24 by tobozo

spacehuhn, 2018-03-24:
I've added a reset sketch https://github.com/spacehuhn/esp8266_deauther/tree/master/Reset_Sketch