SpacehuhnTech/esp8266_deauther #683: Stuck on "starting..."

dodgeboy, 2017-12-31:
I recently purchased a pair of ESP-12E development boards (https://www.amazon.com/gp/product/B010N1SPRK/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1) to use for this purpose.  I had to modify the baud rate in the source to 74880, because with 115200, I receive gibberish on the serial term (regardless of connected baud rate).  Once I modified the baud rate, I was able to see the startup (pasted below).  It gets all the way to "starting..." and then hangs with a red LED.  I never see the SSID, nor can I connect to it as if it were hidden, by specifying the SSID.  I've tried flashing the .bin as well, but since I cannot see the serial output with the .bin, I can't do any debugging.  I suspected it may be my power source, since it seemed to die when the WiFi radio was enabled, but I don't get any success with any available power source.  I'm uncertain whether the code is somehow incompatible with this particular board, or if I am doing something wrong.  I'd appreciate any assistance.

Board:	ESP8266 Dev Module ESP-12E
Using OLED:   NO
Using WebServer: NO (because it doesn't get that far)
Using Serial: YES
Installation mode:  Source
Code/binary version: 1.6
Flash size:  4M
Flash method:	Arduino IDE
Flash Frequency:	80MHz
Upload Speed:	115200
Powered by:  2.4A USB Wall Charger, Powered USB Hub, Battery pack, etc.  No success

Serial output:

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
Settings:
SSID: pwned
SSID length: 5
SSID hidden: 0
password: deauther
password length: 8
channel: 1
Default MAC AP: 62:01:94:70:f5:13
Saved MAC AP: 62:01:94:70:f5:13
MAC AP random: 0
Scan hidden APs: 1
deauth reson: 1
attack timeout: 300
attack packet rate: 10
client scan time: 15
use built-in LED: 1
channel hopping: 0
multiple APs: 0
multiple Attacks: 0
mac change interval: 4
1s beacon interval: 0
LED Pin: 2
Trying to sync the MAC addr with settings
Synced with saved mac addr : 62:01:94:70:f5:13
nameList cleared
Attack LED : OFF

Starting WiFi AP:
SSID     : 'pwned'
Password : 'deauther'
-----------------------------------------------

 generating MACs...done

Starting...

marek26340, 2018-01-05:
You can try this (copied from github.com/martin-ger/esp_wifi_repeater:

Sometimes, especially on old ESP-01s, there is a wrong or non-matching version of "esp_init_data_default.bin" in the flash. If the firmware files from above flash correctly but after reboot you see only garbage on the serial and/or the LED on GPIO2 is flashing rapidly, try to re-initialize this sector: download https://github.com/espressif/ESP8266_NONOS_SDK/raw/master/bin/esp_init_data_default_v08.bin and flash it to 0x7c000 for 512 kB modules (some ESP-01, Sonoff Switch), 0xfc000 for 1 MB modules (most ESP-01), or 0x3fc000 for 4 MB modules (most ESP-12, Wemos D1).

dodgeboy, 2018-01-05:
Thank you for the response!  Unfortunately, that had no change on the behavior of my specific board.  I’m able to see the output of the serial console if I build from source, and change the baudrate (of the code) to 74880.  That’s how I can see it go through it’s startup and get to the “Starting…” message, where it hangs.  

I tried flashing this over top of a bin flash, and a source flash.  It didn’t help either case.  Thank you for trying, though!

From: marek26340
Sent: Friday, January 5, 2018 2:31 AM
To: spacehuhn/esp8266_deauther
Cc: dodgeboy; Author
Subject: Re: [spacehuhn/esp8266_deauther] Stuck on "starting..." (#683)

You can try this (copied from github.com/martin-ger/esp_wifi_repeater:
Sometimes, especially on old ESP-01s, there is a wrong or non-matching version of "esp_init_data_default.bin" in the flash. If the firmware files from above flash correctly but after reboot you see only garbage on the serial and/or the LED on GPIO2 is flashing rapidly, try to re-initialize this sector: download https://github.com/espressif/ESP8266_NONOS_SDK/raw/master/bin/esp_init_data_default_v08.bin and flash it to 0x7c000 for 512 kB modules (some ESP-01, Sonoff Switch), 0xfc000 for 1 MB modules (most ESP-01), or 0x3fc000 for 4 MB modules (most ESP-12, Wemos D1).

dodgeboy, 2018-01-05:
It is definitely something with this particular board, because I can successfully flash it to a Wemos D1, and it works just fine.  I have several of these boards, though.  How can I debug this, to get it working with these boards?  Thanks.

tobozo, 2018-03-24:
see if it is fixed in the next version, reopen if necessary

closing this

closed 2018-03-24 by tobozo