SpacehuhnTech/esp8266_deauther #686: Need help with configuration (reset settings, without logging in through the browser)

Wither2, 2018-01-03:
Hello, please tell me how I can reset the settings without logging into the browser.  I just changed my password and forgot it. As a result, I can not connect to its Wi-Fi network. And I do not know what to do.  I ask you, help.

loskiq, 2018-01-03:
**How do I reset it?**

Method 1: Connect pin 4 (D2 on the NodeMCU) to GND and plug the device in.
Method 2: Connect your device, open up the serial monitor in Arduino, change the baudrate to 115200, type in "reset" and click send.

https://github.com/spacehuhn/esp8266_deauther/wiki/FAQ

Wither2, 2018-01-03:
Sorry, but you can tell me more about 2 variants. I'm just completely new, and I do not understand many things. I'll be infinitely grateful if you help me.

Wither2, 2018-01-03:
On the second option

loskiq, 2018-01-03:
Connect ESP8266 to your computer via USB cable. Then open console client, select com port and baud rate (115200). Type "reset" and click Enter.

Wither2, 2018-01-03:
Tell me please, "open the Windows console client" is it in Arduino or in CMD Windows?

Wither2, 2018-01-03:
And please tell me which module in Arduino you need to select for this board, in order to reset the settings and see all the code as a whole.

[image]
[image]

tobozo, 2018-01-03:
have you tried pushing this button while the device is powered ?

[image]

Wither2, 2018-01-04:
Yes, the device simply restarts.  
Please, let's get to sleep in VKontakte, here is my sramnica: https: //vk.com/xax101001 
Or in any social network convenient for you.

spacehuhn, 2018-03-24:
You can try the new version. The other reset options are explained above.

closed 2018-03-24 by spacehuhn