SpacehuhnTech/esp8266_deauther #687: Encryption option not working

KnifeBoss, 2018-01-03:
### Hardware/Flashing:
Board: ESP8266 
Using OLED:  NO
Using WebServer: YES
Using Serial: ?YES?NO?
Installation mode:  binary
Code/binary version: ?1.5?1.6?
Flash size:  1M
Flash method:	esptool
Flash Frequency: 80Mhz?
Upload Speed: 115200
Powered by:  USB

### Description:
When I select the option to make AP with encryption, and I press Start the AP appear but they are not encrypted (no pass)

What you're expecting:  Im expecting the AP to be encrypted with a password.
What you get instead: Unsecure AP

### Debug Messages:
```
Put the serial output here

marek26340, 2018-01-09:
What code version are you using? I hope that you are using the latest one.

spacehuhn, 2018-01-10:
That's known issue on Apple devices. Will be fixed in the next version.

leperjulien, 2018-02-04:
Same for me of Apple Computer and smartphone :) Do you know what part of the code will be modified in order to fix this ?

Thanks

tobozo, 2018-03-24:
see if it is fixed in the next version, reopen if necessary

closing this

closed 2018-03-24 by tobozo