SpacehuhnTech/esp8266_deauther #689: Pwend network not showing up

Johnsjs232, 2018-01-05:
### Hardware/Flashing:
Board:				Esp8266
Using OLED:  no
Using WebServer: ?YES?NO?
Using Serial: ?YES?NO?
Installation mode:  ?binary?source?
Code/binary version: ?1.5?1.6?
Flash size:  1M
Flash method:	Nodemcu flasher program
Flash Frequency: 40Mhz
Upload Speed: 115200
Powered by: Usb

### Description:
It says that it is complete but it will not give off a wifi signal? I have no idea what to do
What you're expecting:  Idk
What you get instead: ---

### Debug Messages:
```
Put the serial output here

marek26340, 2018-01-05:
Please give us the serial output from the serial monitor in Arduino IDE. Also it would be better if you will fill out the information like "Installation mode: binary/source" properly.

Johnsjs232, 2018-01-11:
where do I find this info

XAS-712, 2018-01-20:
Connect its serial port to you computer. Then run a serial port monitor and open the port.

hmpowerbt, 2018-02-11:
Try adding the network its probably hidden.

tobozo, 2018-03-24:
see if it is fixed in the next version, reopen if necessary

closing this

closed 2018-03-24 by tobozo