SpacehuhnTech/esp8266_deauther #695: 'freedom_outside_cb__t' was not declared in this scope

generaldefence, 2018-01-20:
### Hardware/Flashing:
Board: ESP8266
Using OLED: NO
Using WebServer: NO
Using Serial: NO
Installation mode: Code
Code/binary version: 1.6
Flash size:  ?1M/512K?
Flash method: Arduino IDE?
Flash Frequency: 80Mhz
Upload Speed: 115200
Powered by: USB

### Description:
I have placed all the files in all the right places and edited them correctly as well, but now it has stopped at where freedom_outside_cb__t was not declared in the scope. I have tried to find what to do, but it didn't work. If anyone can help me get this deauther going, it would be of great help.

What you're expecting: To flash my code
What you get instead: Something not declared

### Debug Messages:
```
Arduino: 1.6.10 (Mac OS X), Board: "Adafruit HUZZAH ESP8266, 80 MHz, Serial, 115200, 4M (1M SPIFFS)"

In file included from sketch/Settings.h:12:0,
                 from sketch/APScan.h:9,
                 from sketch/APScan.cpp:1:
/Users/name/Library/Arduino15/packages/esp8266/hardware/esp8266/2.0.0/tools/sdk/include/user_interface.h:452:39: error: 'freedom_outside_cb__t' was not declared in this scope
 int wifi_register_send_pkt_freedom_cb(freedom_outside_cb__t cb);
                                       ^
exit status 1
Error compiling for board Adafruit HUZZAH ESP8266.

XAS-712, 2018-01-20:
Check your library version first. It must be 2.0.

0xCoto, 2018-01-20:
delete self

0xCoto, 2018-01-20:
rm rf

0xCoto, 2018-01-20:
rm rf

0xCoto, 2018-01-20:
rm rf

0xCoto, 2018-01-20:
rm rfrm rfrm rfrm rfrm rfrm rfrm rfrm rfrm rfrm rfrm rfrm rfrm rfrm rfrm rfrm rfrm rfrm rfrm rfrm rfrm rfrm rfrm rfrm rf

tobozo, 2018-03-24:
closing this (see ReadME and FAQ)

closed 2018-03-24 by tobozo