SpacehuhnTech/esp8266_deauther #696: AP doesn't show up after flashing

TheLastOption, 2018-01-20:
### Hardware/Flashing:
Board: ESP8266 
Using OLED:  NO
Using WebServer: NO
Using Serial: NO
Installation mode: binary
Code/binary version: 1.6
Flash size:  1M
Flash method:	NodeMCU flasher
Flash Frequency: 80MHz
Upload Speed:115200
Powered by: USB plugged into wall

### Description:
I tried the Arduino way of uploading the code first and only got a red LED that was always on. So, I tried to just flash it using the bin files and after it's done flashing, I unplug it, plug it into the wall, the blue LED near the top of the board flashes once, then no network shows up on my phone or laptop. Any help?

What you're expecting:  --- The network to show up on my phone
What you get instead: --- nothing

### Debug Messages:
Put the serial output here 
The log in the flasher just says "Note: Program flash success." Literally have no idea what the problem is.

hmpowerbt, 2018-02-11:
Try adding the network because its probably hidden.

tobozo, 2018-03-24:
see if it is fixed in the next version, reopen if necessary

closing this

closed 2018-03-24 by tobozo