SpacehuhnTech/esp8266_deauther #698: Do I need to be in range of the users or router

sam-gurdus, 2018-01-20:
So, if I want to knock everyone off of a network, do I need to be in range of them or just the router?

tobozo, 2018-01-21:
Very good question, you may find the answer by analysing the theory of the attack, and see who's concerned (AP or station) by the management framesyour ESP sends.

If you don't feel like reading the code, you may instead check the first link in the [ReadME](https://github.com/spacehuhn/esp8266_deauther#what-it-is-and-how-it-works) and find out the answer by reading the relevant WikiPedia article.

closed 2018-01-21 by tobozo

spacehuhn, 2018-01-21:
In the current version you need to be in range of both, because in order to attack clients, you need to scan and select their AP too. Next version will only require beeing in range of one of both.