SpacehuhnTech/esp8266_deauther #700: Customize HTML/CSS

sam-gurdus, 2018-01-21:
Is there the possibility to add a customization option in the settings so we can edit the HTML/CSS without having to edit the source and compile it ourselves.

tobozo, 2018-01-21:
you may try the [testing branch](https://github.com/spacehuhn/esp8266_deauther/tree/testing) and edit everything in the /data folder but it requires the Skecth Data Uploader component for the Arduino IDE (see the ReadME) and it's highly unstable.

Otherwise you can wait for the next release, @spacehuhn may reveal the roadmap soon.

XAS-712, 2018-01-21:
It's a good idea. But the length of byte arrays is fixed. You may not change it successfully.

sam-gurdus, 2018-01-22:
So is the new release a complete overhaul? Do we know when?

tobozo, 2018-01-22:
could be serverless version as most problems come from the web server

sam-gurdus, 2018-01-22:
What does that mean? You mean it doesn't have a web interface or what?

tobozo, 2018-01-22:
I said it *could*, but in reality I just hope it won't have a web interface. 
Most issues in this version go away as soon as you disable the webserver so that would be fair.

sam-gurdus, 2018-01-22:
How would you control it then?

tobozo, 2018-03-24:
new approach to customizing web content is introduced by the new release, closing this

closed 2018-03-24 by tobozo

spacehuhn, 2018-03-24:
Tutorials and infos on how to modify it follow soon, keep an eye on the wiki.
In the meantime:
- edit the stuff in the web_interface folder
- convert it with the webConverter.py in utils
- upload it using the SPIFFS arduino addon (google is your friend)