SpacehuhnTech/esp8266_deauther #702: DEAUTH attack always uses broadcast address

HardResetSolution, 2018-01-22:
### Hardware/Flashing:
Board: Deauther OLED V1.5 ESP8266+
Using OLED: YES
Using WebServer: NO
Using Serial: YES
Installation mode:  source
Code/binary version: 1.6
Flash size:  512K
Flash method: Arduino IDE
Flash Frequency: 40Mhz
Upload Speed: 115200
Powered by:  USB

### Description:
I noticed, that all DEAUTH attacks I'm doing, are always done with the broadcast address "FF:FF:FF:FF:FF:FF" as client's MAC-address. It seems like clientScan.results is always empty.
clientScan.packetSniffer(buf, len) seems to be never called, as debug output is never generated from that method.

HardResetSolution, 2018-01-23:
Well, I found the reason for only sending DEAUTH-Request to/form broadcast address: The method "void ClientScan::start(int _time)" is only called by a request from the webserver. As I'm not using the webserver, clients get never scanned. Furthermore in "Attack::run()" there is an if statement that leeds to send DEAUTHs only to clients that are selected.

Is that the expected behavior?
Why not scan and use the MAC-addresses of all the connected clients instead of using the broadcast address?
I tested the DEAUTHER against several clients. All linux clients got disconnected by the DEAUTHER, but most windows clients were resistant against the DEAUTHER.

spacehuhn, 2018-01-23:
No worries I'm working on it 😃  I already got a good a reliable scan working for the next verison.
The reason it's so *limited* in the current version, is because the display interface was a quick and dirty addon. I'm rewriting the whole codebase.

tobozo, 2018-03-24:
see if it is fixed in the next version, reopen if necessary

closing this

closed 2018-03-24 by tobozo