SpacehuhnTech/esp8266_deauther #704: Try to remove deauther but AP is still there

RagnarL18, 2018-01-27:
Dear,

I try to delete deauther completely from my NodeMCU v1 but everytime if I flash the NodeMCU Firmware or I try to erase the flash the AP from the deauther is still there and I can connect to it. I cannot open the deauther web interface but the WiFi network is still there. How can I remiove deauther completely ???

tobozo, 2018-01-27:
does it persist after you power-cycle both the ESP and your device (mobile or PC?)

RagnarL18, 2018-01-28:
Yes

tobozo, 2018-01-29:
can you describe how you did the flash erase ? did you also erase the data ?
please give details

also check [this site](http://www.pratikpanda.com/completely-format-erase-esp8266-flash-memory/), especially the part where it says:

```
If you want to erase entire flash, for example of an ESP-12 E module, which has a 4MB 
flash memory chip, you need to upload the blank_1MB.bin file to 0x000000, 0x100000, 
0x200000 and 0x300000. This will put 0xFF in the range 0x000000-0x3FFFFF. 
Which is same as clearing the entire flash!
```

marek26340, 2018-01-30:
This is what I found in a quick google search...

"If you want to factory erase based upon a button push -- and the factory erase function clears out the SSID values, then it looks like you need to write a little function that will clear the following address on most hardware: 0x7E000 -- around 128 bytes looks like the values you want to clear. On my flash, it is all FF's before a valid SSID is stored there. If that is not your address you want to find the right area. But anyway, you can check if its correct by reading from that area using esptool. No need to delete the entire flash if you want to nuke those settings."

DmitryXO, 2018-02-27:
I used esptool to upload on wemos d1promini
`esptool.py --port /dev/tty.SLAB_USBtoUART -b 115200 write_flash --flash_size 4MB 0x0 esp8266_deauther_1mb.bin`

then to remove existing ssid, eeprom doesn't work for me
`esptool.py --port /dev/tty.SLAB_USBtoUART -b 115200 write_flash --flash_size 4MB 0x0 blank_1MB.bin` 
blank_1MB.bin is a simple 1mb empty file, goggle for it or make it by yourself.

tobozo, 2018-03-24:
see if it is fixed in the next version, reopen if necessary

closing this

closed 2018-03-24 by tobozo

spacehuhn, 2018-03-24:
There is a reset sketch now: https://github.com/spacehuhn/esp8266_deauther/tree/master/Reset_Sketch

rektbyme, 2018-03-27:
here you have a blank 1mb file :+1: 
[blank_1MB.zip](https://github.com/spacehuhn/esp8266_deauther/files/1850030/blank_1MB.zip)

rektbyme, 2018-03-27:
ill do a summary of all of this for the next who encounter this problem, you will erase your arduino shit like doing a factory reset: 
1. you need to install python, then pip install esptool
2. connect your arduino shit to your pc
3. look in what com port you have the arduino shit (example: COM5)
4. move to the directory of scripts inside your python installation folder and if you have esptool already installed you can proceed opening a console prompt in that scripts folder
5. in that prompt folder on the path of scripts you can run: "python esptool.py -p com5 erase_flash"

There is a good tutorial about how to install python and the stuff i said before, google this: "NodeMCU V2 ESP8266 ESP-12E Flashing Update Firmware using esptool.py Python in windows Tutorial 2" is a good youtube video if you arent familiarized with python :)