SpacehuhnTech/esp8266_deauther #709: I Upload it->Red light constantly on->Nothing happens

James20002, 2018-02-05:
### Hardware/Flashing:
Board:ESP8266 ESP12-E?
Using OLED: NO
Using WebServer: NO
Using Serial: YES
Installation mode:  BINARY
Code/binary version: 1.6
Flash size:  1M
Flash method:	IDE
Flash Frequency:40Mhz
Upload Speed:On back it says 9600 but I have tried it and 115200
Powered by: USB

### Description:
I have tried uploading and erasing and changing settings but nothing has happened but a red light stays on, blue light blinks once and goes dead, and red and blue lights blink back and forth...not sure what to do other than to keep switching settings and try again but I just don't know...please help.

What you're expecting:  ---idk
What you get instead: ---idk

marek26340, 2018-02-05:
Hmm... How could you flash the ESP with binary file through Arduino IDE?
Anyways, you can also try this: Get a ESP8266 flash tool and this .bin file: 

> https://github.com/espressif/ESP8266_NONOS_SDK/blob/master/bin/esp_init_data_default_v08.bin

and flash it to 0x7c000 for 512 kB modules (some ESP-01, Sonoff Switch), 0xfc000 for 1 MB modules (most ESP-01), or 0x3fc000 for 4 MB modules (most ESP-12, Wemos D1).

spacehuhn, 2018-03-04:
What is even the problem here despite blinky LEDs when uploading?

spacehuhn, 2018-03-24:
No answer... closing this.

closed 2018-03-24 by spacehuhn