SpacehuhnTech/esp8266_deauther #713: Attackinfo.json

hmpowerbt, 2018-02-11:
Everytime I test the deauth option I get an error that says "error loading attackinfo.json". If anyone knows how to fix this comment the steps please. Thanks!:blush:

hmpowerbt, 2018-02-11:
Somebody asked me what version Im running  (Im currently running v1.6)

hmpowerbt, 2018-02-11:
My board is (Esp8266 NodeMCU Amica)

Dev90564, 2018-02-11:
I have the same problem too! I also have the same board and same version running. This also happend on the nodemcu lolin too!! Everytime this happens I must unplug my esp8266 ans replug it.

hmpowerbt, 2018-02-11:
@Dev90564 Good to know im not the only one. I thought my board was ruined 😂😁.

Pow674, 2018-02-11:
I have this problem too and I want to know how to fix it. I also have to unplug and replug. Thats my only problem too!

hmpowerbt, 2018-02-11:
@pow674 thats my only problem too

FlowBot123, 2018-02-11:
I think you must change your band. But i dont know how to do that?? Please help😡

FlowBot123, 2018-02-11:
My deauther is not working correctly too😕

hmpowerbt, 2018-02-11:
@flowbot123 just search up how to change the band:grin:

hmpowerbt, 2018-02-11:
@flowbot123 please email me. My email is in my profile description

hmpowerbt, 2018-02-11:
@flowbot123 i have changed my channel and it still says error loading attackinfo.json. i did this to 3 other boards too!

Dev90564, 2018-02-11:
Ok... I dont know what else to do. I have tried everything and i still get this error.

Dev90564, 2018-02-11:
Ill check my channel too

hmpowerbt, 2018-02-11:
Did u change ur band @dev90564

Dev90564, 2018-02-11:
Yes i have!

hmpowerbt, 2018-02-11:
@dev90564 how did u do that???

Dev90564, 2018-02-11:
Same as channel

hmpowerbt, 2018-02-11:
@dev90564 ok....????

MoonPie000, 2018-02-11:
Same problem too.. just reflash the old versions.. if they show the same problem then idk what else u can do.

hmpowerbt, 2018-02-11:
Did not work 😕 @moonpie000

Pow674, 2018-02-11:
Didn't work for me too!! @MoonPie000

MoonPie000, 2018-02-11:
Oh well srry

hmpowerbt, 2018-02-11:
Any other methods

hmpowerbt, 2018-02-11:
Oh yeah @dev90564 channel is not the same as band!!

hmpowerbt, 2018-02-12:
@pow674 did not work for me either

MoonPie000, 2018-02-12:
Flash back to 1.0 then 1.1, 1.2, 1.3, 1.4, 1.5, 1.6 and see if it works

hmpowerbt, 2018-02-12:
@moonpie000 ok ill try that

hmpowerbt, 2018-02-13:
Do not try what @moonpie000 suggested. It ruined my esp8266. I dont really care because i have a crap ton of these thing.

Pow674, 2018-02-13:
I was about to try it! Thankfully I was lazy enough not to do it..... I also think i found another way to remove the error. RESET the esp8266 and reflash 1.6

hmpowerbt, 2018-02-13:
@pow674 i have tried this already

hmpowerbt, 2018-02-13:
@pow674 ur method does not work unfortunately.

marek26340, 2018-02-13:
This error can occur when you will try to attack an AP that's on a different frequency than the deauther's AP's frequency. (for ex. Your deauther is set to be on channel 1. You want to deauth an AP that is on channel 11. You start the attack and then instantly the deauther will switch channels and your phone is now unable to communicate to the ESP, and that's when the error "error loading attackinfo.json" comes. Now you need to disconnect from the deauther's AP and reconnect to it, after that reload the page and the error should be gone.)

Now some more details attackinfo.json
attackinfo.json shows in the browser which attack is running and it's packetrate. I think that it refreshes every second when in the Attacks tab. If the browser is unable to refresh this file, it displays that error. Why is it unable to refresh this file? 
First, the ESP may have switched channels when you started an attack.
Second, the deauther firmware itself might have encountered an error and crashed.
Third, the ESP is damaged or the deauther firmware has been corrupted or flashed incorrectly.

hmpowerbt, 2018-02-14:
@marek26340 thank you very much. Your methods have helped alot and have been successful! 😁

closed 2018-02-14 by hmpowerbt

VainXploits, 2023-11-01:
how do I check what channel the ESP is on?