SpacehuhnTech/esp8266_deauther #714: Unable to find pwned network

EyesightTechnology, 2018-02-13:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware/Flashing:
Board: NodeMCU from here: https://www.amazon.com/gp/product/B010O1G1ES/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1
Using OLED:   No
Using WebServer: I am using the web interface, so I think so
Using Serial: No
Installation mode:  binary
Code/binary version: 1.6
Flash size: 512K
Flash method:	nodemcu-flasher
Flash Frequency:	80Mhz
Upload Speed:	921600
Powered by:  USB

### Description:
Well, I started up the module after flashing it as per specifications above.  Once flashing was complete, I unplugged my NodeMCU and plugged it back in. The red LED was now solidly on. (When I used the 1MB version, the red LED did not light up. Is this a good or a bad sign?)

What you're expecting: 
In my network lists on my computer, there would be a new network called "pwned"
What you get instead:
No new network was found.

### Debug Messages:
I have not connected the module to serial.

Any help would be greatly appreciated! Thanks!

hmpowerbt, 2018-02-14:
Try adding pwned as a hidden network.                                             Network Name: Pwned                     
Security: WPA2                                   Password: deauther                          If that does not work. Reset and reflash your NodeMCU.

EyesightTechnology, 2018-02-14:
Ok. So here's what I did. (Didn't work!!!)

1. Click on Hidden Network in my network menu
2. It asks me for an SSID/name. I enter "pwned"
3. For quite a while, I get a loading bar labeled "Checking network requirements" or something similar
4. It then prompts me for a password. I enter "deauther."
5. It loads for a while
6. A message appears which says "Can't connect to network" or "Could not be joined" or something similar.

The fact that it takes so long to load (30-60 seconds) makes me believe the computer is detecting SOMETHING. Right?

hmpowerbt, 2018-02-14:
Did you select WPA/WPA2/FT PSK

hmpowerbt, 2018-02-14:
Try plugging it in your pc/desktop. If that does not work try using a power bank.

EyesightTechnology, 2018-02-14:
I tried it in my Windows Laptop, my Mac Laptop, and in a wall plug. 

Could something be wrong with my module? **Is the red LED supposed to be on during operation?**

hmpowerbt, 2018-02-14:
Yes, it's supposed to be on. What program are you using to flash youd module. NodeMCU flasher or arduino?

EyesightTechnology, 2018-02-14:
I used NodeMCU flasher. You can see all my upload specs in the original post.

hmpowerbt, 2018-02-15:
Try Reflashing it. If you have and still have the problem try using a 5V charger to power it. If you have already done that then try adding it as a hidden network again.

hmpowerbt, 2018-02-17:
Its supposed to flash blue not red.

EyesightTechnology, 2018-02-18:
The module has two LESs, from what I can see.

When I flash the code, the blue LED blinks rapidly.

When the code is done, the red LED goes solidly on.

PwnKitteh, 2018-02-22:
The blue LED is tx on the nodemcu, which means that when the NodeMCU is sending data, that LED will flash. No led flash, no data is sent. Try flashing using esptool.py
the flash location is 0x000 . It's CLI, but I can confirm it works. I haven't used esp8266 flasher, as I don't run windows.

ALoGeNo, 2018-02-22:
Hi, trying to help, I guess that here the problem is that the device has just "Flash size: 512K" and we must edit the mac data file (I did read the README after flash the [512K .bin 1.6 version](https://github.com/spacehuhn/esp8266_deauther/releases/download/v.1.6/esp8266_deauther_512kb.bin), and voila!, my esp8266 (esp-201) finaly works.
Excuse me if I'm wrong.

Thank you for sharing, good job guys!.

Cheers.

tobozo, 2018-03-24:
see if it is fixed in the next version, reopen if necessary

closing this

closed 2018-03-24 by tobozo

juliusfrost, 2018-04-08:
I'm getting the same issue...

rogaha, 2018-05-12:
same issue here

tobozo, 2018-05-12:
@rogaha same answer here: please open a new issue and be verbose about what you expect, how you proceed, a what result you get.
Also describe your environment with as many details as possible.

rogaha, 2018-05-12:
@tobozo, sure, np. here we go:https://github.com/spacehuhn/esp8266_deauther/issues/858

Sabarivasan78, 2021-08-05:
> Hi, trying to help, I guess that here the problem is that the device has just "Flash size: 512K" and we must edit the mac data file (I did read the README after flash the [512K .bin 1.6 [...]

It works thanks a lot

Sabarivasan78, 2021-08-05:
> ### Hardware/Flashing:
> Board: NodeMCU from here: https://www.amazon.com/gp/product/B010O1G1ES/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1
> Using OLED: No
> Using WebServer: I am using the web interface, so I think so
> Using Serial: No
> Installation mode: binary
> Code/binary version: 1.6
> Flash size: 512K
> Flash method: nodemcu-flasher
> Flash Frequency: 80Mhz
> Upload Speed: 921600
> Powered by: USB
> 
> ### Description:
> Well, I started up the module after flashing it as per specifications above. Once flashing was complete, I unplugged my NodeMCU and plugged it back in. The red LED was now solidly on. (When I used the 1MB version, the red LED did not light up. Is this a good or a bad sign?)
> 
> What you're expecting:
> In my network lists on my computer, there would be a new network called "pwned"
> What you get instead:
> No new network was found.
> 
> Please note, I am using Windows 10. Could it be that Windows 10 is hiding the network? I see two "Hidden Network"s in my wifi options
> 
> ### Debug Messages:
> I have not connected the module to serial.
> 
> Any help would be greatly appreciated! Thanks!

I think you not selected the bin file.select the bin file and flash it again.