SpacehuhnTech/esp8266_deauther #716: Able to dearth 3G/4g connections?

EPiCxRIOTZZ, 2018-02-18:
Is there a way of being able to deauth 3G/4g LTE connections

tobozo, 2018-02-18:
yep, usually you switch your phone to plane mode and that does it

marek26340, 2018-02-21:
Well, now the reality. NO, you can't disconnect all mobile users from the mobile network, not even one single user. The ESP8266 operates in the ISM band, that is 2.4~2.5GHz. The mobile network operates at 800, 850, 900, 1800, 1900, 2100, 2600 MHz. By this I want to explain that mobile and WiFi frequencies are different...

And also, **deauthing or jamming any mobile network is highly illegal!**

jLynx, 2018-03-06:
That's where you are wrong, my friend who has a mate who's brothers friend has done it with an ESP. He got sent to jail though, but the good news is he broke out recently and is still "dearth"ing all the 3g and 4g networks

closed 2018-03-06 by jLynx

marek26340, 2018-03-06:
But just how?! This is impossible with an ESP8266! wtf

I think that you are lying. I think that this is just a hoax...

jLynx, 2018-03-06:
Nope fully possible, as I said my sisters friends bothers sons friend has done it

spacehuhn, 2018-03-06:
Yeah man don't you believe jlynx sisters friends brothers sons friend? He has done it!

spacehuhn, 2018-03-06:
[image]

marek26340, 2018-03-06:
omg okay... then what's the code that I have to upload to my ESP? Or what do I have to do?
I'd really like to deauth the 4G networks too! I would really like to hear the fear from all the 300 kids at our school that the internet is not working! xD that would be really funny lol

deantonious, 2018-03-06:
Once you open the corner, look for the smd pad inside and solder the 3/4G antenna to it using some hot air or a thin soldering tip (works on the ESP32 too):
 
[image]
[image]

Hope this helps you :)

r0073rr0r, 2018-03-06:
troll to the perfection :D

jLynx, 2018-03-06:
@marek26340  let me know if you need any help with setting it up. But please make a nicer hole than @deantonious . Yes his works, but doesnt look nice at all to the eye

marek26340, 2018-03-07:
ehm... it works on an esp32 too?

jLynx, 2018-03-07:
Yeah it does which is many people don't realise

Schokobecher, 2020-01-08:
I can vouch for this working!