SpacehuhnTech/esp8266_deauther #717: Powering esp8266 nodemcu with 9v through vin gives a shocking result!

Booooozo, 2018-02-19:
So I have an amica esp8266 nodemcu which I flashed the deauther program on. I love it and it works fine. But I found out that powering the nodemcu through vin port with a 9v battery sends deauthentication packets around my whole neighborhood. I tested this by simply browsing on my iPad and then powering up the nodemcu through vin and immediately my iPad disconnected from my WiFi and the same happened when I was using a mobile hotspot. It’s really and also my nodemcu led does not light up when this is happening. Do you guys experience this same thing?

spacehuhn, 2018-02-19:
Black Magic

kaosproject, 2018-02-20:
Vin and G ?

Booooozo, 2018-02-20:
Yes no external antenna I connected it to VIN and G

abdullahseba, 2018-02-20:
Yeah its simple really. `9V x 20dVm = 180VdBm` So that means you get more packets per meter which means there is more for each device. `180VdBm ÷ 1000mm = 0.18 packets rms` while with 5v all you get is 0.04 `40VdBm ÷ 1000mm = 0.04 packets rms`.

tobozo, 2018-02-20:
neat ! just add a pot

marek26340, 2018-02-22:
@abdullahseba wtf I don't understand.................

digmorepaka, 2018-03-04:
@marek26340 quick mafs

abdullahseba, 2018-03-05:
^^

marek26340, 2018-03-19:
quick, but weird mafs

spacehuhn, 2018-03-24:
Was fun but closing this now.

closed 2018-03-24 by spacehuhn