SpacehuhnTech/esp8266_deauther #718: esp8285

loskiq, 2018-02-20:
Is there support for ESP8285?

marek26340, 2018-02-21:
not now, but spacehuhn should be able to tell you more info.

tobozo, 2018-02-21:
Apart from the boards.txt missing the ESP8285 entry (and thus making it impossible to flash from the Arduino IDE /w sdk 2.0.0) what is preventing the deauther code from working with this model ?

marek26340, 2018-02-21:
idk :D

tarzan212, 2018-02-28:
The exploit has not been fixed in any new wifi ships  for the ESPs as far as I know ? 
Isn't it working on any other board ? (Not even the esp32 ?)

spacehuhn, 2018-03-24:
With @tobozo new modified ESP8266 SDK for Arduino 
https://github.com/tobozo/Arduino
you can select the ESP8285 as a board and flash it. I tested it and it works 😃

tobozo, 2018-03-24:
closing this as resolved

closed 2018-03-24 by tobozo