SpacehuhnTech/esp8266_deauther #719: What button do I click to auto deauth all routers in range?

Booooozo, 2018-02-20:
So I’ve heard that by clicking the reset or flash button on my amica esp8266 nodemcu will deauth all routers in range i have tried it but it doesn’t work. Is this true that by clicking the reset or flash button will deauth everyone or does my board not work with this feature?

Source from where I heard it: An open issue in your project which says something like “Stuck on Starting....”

spacehuhn, 2018-02-20:
I heared that if you leave it in water for a minute it will actually start jamming Cabrios frequentiert  due to the freed electrons on the metal finish.

spacehuhn, 2018-02-20:
sorry wrong button

Booooozo, 2018-02-21:
Nice joke. Haha

andixum, 2018-02-21:
put it in your toaster to warm up the signal a bit. That should make it uber powerful and will deauth everything.

Booooozo, 2018-02-21:
Hahahahaha!

closed 2018-02-21 by Booooozo