SpacehuhnTech/esp8266_deauther #720: ESP8266_deauther - Automatic Deauthenticating

veryverynoobteamates101, 2018-02-23:
is automatic deauth is already added as a feature?
Meaning, instead of selecting one of the wifi devices to attack in deauth mode, you'll just have to use deauth without selecting one AP on scan section and without scanning as it automatically deauths every wifi devices that get to its radius or that reached its signal range.

tobozo, 2018-02-23:
good things come in time to those who can wait

veryverynoobteamates101, 2018-02-24:
As what I've heard, It is impossible to use auto deauth in all devices that is in range/radius because if the APs are unknown then there is no means of deauth anyway that's why we need to scan in order to do deauths.

Just my idea, what if we use auto scanning and auto APs selection in every (seconds) followed by auto deauth all, would it make it possible?

tobozo, 2018-02-24:
I'm quite sure this question is in everyone's mind, but there's only one chicken to reply to your answer and he's probably busy doing some chickeny stuff :chicken:

spacehuhn, 2018-02-24:
🐔

killergeek, 2018-02-24:
chicken

vuong244, 2018-02-25:
chicken

EmiSchurr, 2018-02-25:
Huhn

XAS-712, 2018-02-26:
hhhh

vuong244, 2018-02-26:
Added features "deauth detector" for OLED https://youtu.be/8PjpZZWxiUQ

tobozo, 2018-02-26:
@vuong244 youpi ! first deauth tutorial in Canadian French 👍

spacehuhn, 2018-03-24:
To answer the question: It is possible with the 2.0 version. There are some notes about doingby writing a script in the wiki. It's not beginner friendly and I don't intent do make it that way because everyone would just abuse it and annoy everyone. This project is for testing.
So yeah if you want to use it, you can do with the new version but I won't do any noob tutorials on it.

closed 2018-03-24 by spacehuhn