SpacehuhnTech/esp8266_deauther #721: hotspot not showing up with esp01

Soumojit28, 2018-02-24:
Hardware/Flashing:
Board: ESP01	
Using OLED:   NO
Using WebServer: no
Using Serial: no
Installation mode:  source
Code/binary version: ?1.6?
Flash size:  Dont know
Flash method:	esp8266 flasher
Flash Frequency 40mhz
Upload Speed 115200
Powered by: from anoter node mcu

### Description:
I hooked up the esp 01 module with a another nodemcu set as a programmer and flash the 1mb bin into the esp 01 module its successfully flashed in the esp01 after that when I run the esp 01 from a 3.3v not hotspot showing I also tried with 512kb flash but still it did not work please help

### Debug Messages:
```
Put the serial output here

marek26340, 2018-02-27:
I think that you really need an FTDI serial programmer, like this one: [https://www.ebay.com/itm/FT232RL-3-3V-5-5V-FTDI-USB-to-TTL-Serial-Adapter-Module-for-Arduino-Mini-Port/381374421597?epid=502148532&hash=item58cbafda5d:g:jk8AAOSwrklVMjIp](url)

Soumojit28, 2018-02-27:
I already ordered one, so You think programming with this could solve the problem ?

marek26340, 2018-02-27:
http://forum.arduino.cc/index.php?topic=348160.0

Here is the answer why it might not be working. How did you connect that ESP to your arduino board?

Soumojit28, 2018-02-27:
I use another node mcu to program my esp01

spacehuhn, 2018-03-24:
I hope you were able to solve the issue.
I'm closing it because it's unrelated to this project. The described problem is a setup/hardware problem.

closed 2018-03-24 by spacehuhn