SpacehuhnTech/esp8266_deauther #722: Beacon list gone after restart

EmiSchurr, 2018-02-25:
### Hardware/Flashing:
Board: NodeMCU ESP8266 AI-Thinker
Using OLED:   NO
Using WebServer: YES
Using Serial: NO
Installation mode:  binary
Code/binary version: 1.6
Flash size:  1M
Flash method: NodeMCU-flasher						
Flash Frequency:	40Mhz
Upload Speed: 9600
Powered by: 5V

### Description:
Beacon list disappears after rebooting device. Beacons still appear when starting 'attack' and can still be added but not individually deleted

XAS-712, 2018-02-26:
The list has been saved in memory. If you restart the board, any data in memory will lost.

marek26340, 2018-02-27:
If you want the ESP to remember the beacon list, there is a button to save the list at the bottom of the page.

EmiSchurr, 2018-02-27:
It has eben saved because upon starting the attack all beacons are shown by every device in range. Adding an Beacon also works, just the list on the web page isn’t shown.

marek26340, 2018-02-27:
Does the serial monitor tell us anything useful? (Connect to pc, make a beacon list, save it, reset esp, try to read the beacon list)

spacehuhn, 2018-03-24:
You have to click save. Otherwise it won't remember them. The list is always in the RAM but if you don't save them before restart, they will get lost.

closed 2018-03-24 by spacehuhn