SpacehuhnTech/esp8266_deauther #726: Not work anymore

Cyberdade, 2018-03-04:
### Hardware/Flashing:
Board:				ESP8266 dev module
Using OLED:   NO
Using WebServer: ?
Using Serial: ?
Installation mode:  binary
Code/binary version: 1.5
Flash size:  ?
Flash method:							esptool
Flash Frequency:					40Mhz
Upload Speed:						115200
Powered by:  Battery

### Description:

Hi, I'm again here.
I found a problem:
When I flash my esp it work, but when I go to the settings and change the Mac to 00:00:00:00:00:00
I can't connect to my esp and if I try to reboot/reflash it I can't find it.
I gave 5€ to the Chineses :'-(
Can anyone help me?

hmpowerbt, 2018-03-04:
Get you board and power it on... then hold the rst button for atleast 5-10 seconds. This will reset the module and erase everything off of. Then reflash the deauther and everything should be back to normal.. hopefully I helped. 😁

hmpowerbt, 2018-03-04:
Get you board and power it on... then hold the rst button for atleast 5-10 seconds. This will reset the module and erase everything off of it. Then reflash the deauther and everything should be back to normal.. hopefully I helped. 😁

Cyberdade, 2018-03-04:
Still not work :-(

spacehuhn, 2018-03-04:
The mac address you gave it is invalid.  
My tip: don't play around with settings if you don't know how to fix it when it breaks!

Reflash your device or see FAQ on how to reset: https://github.com/spacehuhn/esp8266_deauther/wiki/FAQ

Cyberdade, 2018-03-04:
Thanks again, it work!

closed 2018-03-04 by Cyberdade