SpacehuhnTech/esp8266_deauther #728: Deauther implementation on MSP430

LazzariPaolo, 2018-03-04:
Has anyone ever implemented the deauther for the ESP8266 on an MSP430?

killergeek, 2018-03-04:
you know that the msp430 does not have wifi.

spacehuhn, 2018-03-04:
https://en.wikipedia.org/wiki/TI_MSP430
LOL

LazzariPaolo, 2018-03-04:
@killergeek I am aware of that yes.
@spacehuhn 

I am using the UART on the MSP for the configuration and setup of the 8266 wifi module. 
This project and others use Arduino and relevant libraries. 

I wish to implement the deauth attack by sending the deauth packet from the MSP to the ESP so that it can then be broadcast to certain addresses. 
As far as I know, only these AT commands are available for me to use.
https://www.espressif.com/sites/default/files/documentation/4a-esp8266_at_instruction_set_en.pdf

Do you guys have any suggestions?
Thank you.

spacehuhn, 2018-03-04:
Ah ok now it makes more sense! :D
I would say wait on the next release (you can follow my updates on it on twitter @spacehuhn)
because in v2.0 I added a serial command line interface with lot of commands that could do just that.

LazzariPaolo, 2018-03-04:
That's fantastic, cheers! My bad, my open post was very vague indeed.
Is there any way you could tell me what the serial construction of a deauth attack looks like? I am working on this as a project for university and the deadline is tight! I am working on a wifi controller toy car and will take part in a competition. The deauth attack would mean the other students wouldn't even be able to make their car start! 

Thank you in advance.

spacehuhn, 2018-03-24:
Closing this issue because we discussed this on discord.
LazzariPaolo was using an ESP8266 with the AT command firmware and wanted to know if it could be used to send deauth frames. Not possible, because you need a modified firmware for that. It would be possible to modify the AT command firmware but I don't intent to do that for now.

closed 2018-03-24 by spacehuhn