SpacehuhnTech/esp8266_deauther #729: Help the novice...

raphael613, 2018-03-06:
### Hardware/Flashing:
Board:				I think the NOdeMCU v1.0
Using OLED:   ?YES?NO?
Using WebServer: ?YES?NO?
Using Serial: ?YES?NO?
Installation mode:  ?binary?source?
Code/binary version: ?1.5?1.6?
Flash size:  ?1M/512K?
Flash method:	Arduino IDE 1.8.2
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?
Powered by:  ?USB?Battery?
[image]

### Description:
Describe your problem here, provide as many details as possible.
Make sure you read the FAQ first, incomplete bug reports will be CLOSED without notice.

What you're expecting:  to use it
What you get instead: ---

### Debug Messages:
```
"17 Depending on your board you may have to adjust the Tools > Board > Flash Frequency and the Tools > Board > Flash Size. I use a 160MHz flash frequency and a 4M (3M SPIFFS) flash size."
This part is my problem i do no know how to identify my board.
So all fields up i do not know how to fill them 
Just I am using Arduino IDE and I have a NodeMCU v1.0 ithink...not sure...
Thanks in advance.

13r1ckz, 2018-03-06:
what is actually the question/problem because this is just copy paste.

closed 2018-03-06 by jLynx

raphael613, 2018-03-06:
Sorry I did no fill my first issue right
I get nothing
Everything right until uploading
I copied here the error messages I get while uploading
Arduino: 1.8.2 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 160 MHz, Serial, 9600, 4M (3M SPIFFS)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Rephael\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Rephael\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Rephael\Documents\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:CpuFrequency=160,UploadTool=esptool,UploadSpeed=9600,FlashSize=4M3M -ide-version=10802 -build-path C:\Users\Rephael\AppData\Local\Temp\arduino_build_107530 -warnings=all -build-cache C:\Users\Rephael\AppData\Local\Temp\arduino_cache_497780 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Rephael\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=C:\Users\Rephael\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -prefs=runtime.tools.esptool.path=C:\Users\Rephael\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.6 -verbose C:\Users\Rephael\Desktop\USB RUBBER DUCKY\esp8266_deauther-master\esp8266_deauther-master\esp8266_deauther\esp8266_deauther.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Rephael\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Rephael\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Rephael\Documents\Arduino\libraries -fqbn=esp8266:esp8266:nodemcuv2:CpuFrequency=160,UploadTool=esptool,UploadSpeed=9600,FlashSize=4M3M -ide-version=10802 -build-path C:\Users\Rephael\AppData\Local\Temp\arduino_build_107530 -warnings=all -build-cache C:\Users\Rephael\AppData\Local\Temp\arduino_cache_497780 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Rephael\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=C:\Users\Rephael\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -prefs=runtime.tools.esptool.path=C:\Users\Rephael\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.6 -verbose C:\Users\Rephael\Desktop\USB RUBBER DUCKY\esp8266_deauther-master\esp8266_deauther-master\esp8266_deauther\esp8266_deauther.ino
Using board 'nodemcuv2' from platform in folder: C:\Users\Rephael\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0
Using core 'esp8266' from platform in folder: C:\Users\Rephael\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0
Detecting libraries used...
"C:\Users\Rephael\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Rephael\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=160000000L -DARDUINO=10802 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266 "-IC:\Users\Rephael\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266" "-IC:\Users\Rephael\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\variants\nodemcu" "C:\Users\Rephael\AppData\Local\Temp\arduino_build_107530\sketch\esp8266_deauther.ino.cpp" -o "nul"
"C:\Users\Rephael\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Rephael\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=160000000L -DARDUINO=10802 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266 "-IC:\Users\Rephael\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266" "-IC:\Users\Rephael\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\variants\nodemcu" "-IC:\Users\Rephael\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\libraries\ESP8266WiFi\src" "C:\Users\Rephael\AppData\Local\Temp\arduino_build_107530\sketch\esp8266_deauther.ino.cpp" -o "nul"
"C:\Users\Rephael\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Rephael\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0/tools/sdk/include" -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=160000000L -DARDUINO=10802 -DARDUINO_ESP8266_ESP12 -DARDUINO_ARCH_ESP8266  -DESP8266 "-IC:\Users\Rephael\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0. [... cut ...]
	espcomm_send_command: sending command payload
	setting serial port timeouts to 10000 ms
	setting serial port timeouts to 1000 ms
	espcomm_send_command: receiving 2 bytes of data
	writing flash
.....................................................................................................................................................................................................................................................................An error occurred while uploading the sketch
.

spacehuhn, 2018-03-06:
https://www.youtube.com/watch?v=53zkBvL4ZB4

13r1ckz, 2018-03-06:
to mutch, no one is going to read it. read it your self and make a direct question out of it. or make use of the other option like the bin file.