SpacehuhnTech/esp8266_deauther #731: esp32 port

Stavrosfil, 2018-03-14:
Sorry if this is not the place to write this, but i admire your project and i only have an esp32 available, could i somehow use the project there? Thanks in advance!

tobozo, 2018-03-14:
currently not with Arduino IDE, but you can experiment packet injection with the ESP-IDF
https://gist.github.com/brainstorm/24e843ae0295ee1e41dff47c5b43a02c

closed 2018-03-24 by spacehuhn

spacehuhn, 2021-12-12:
Hi! There are many issues on this repo about ESP32 support, so I published a blog post to bring some clarity to this topic: https://blog.spacehuhn.com/esp32-deauther/ 
TL;DR: It's made for the ESP8266, and I have no plans to port this application to the ESP32. But it's open-source, so you're free to do it yourself if you want. I appreciate your understanding.