SpacehuhnTech/esp8266_deauther #733: Deauth doesn't work

avanguard6, 2018-03-15:
Deauth attack suddenly stopped working. Deauth shows 20pkt/s nothing happens. But other attacks are working. I have purchased new one but it also doesn't work.
### Hardware/Flashing:
Board: ESP8266 https://ru.aliexpress.com/item/1pcs-Wireless-module-NodeMcu-Lua-WIFI-Internet-of-Things-development-board-based-ESP8266-CP2102-with-pcb/32801961146.html?aff_platform=link-c-tool&cpt=1521128306960&sk=YFAI6ub&aff_trace_key=07cc03778eee49f4b2db21a203a65812-1521128306960-03437-YFAI6ub&terminal_id=50d88890d2144c09ac426b2f14e41c48
Using OLED:   YES
Using WebServer: yES
Installation mode:  binary
Code/binary version: 1.6

tobozo, 2018-03-15:
what device are you trying to deauth ? maybe it isn't vulnerable to deauth attacks

avanguard6, 2018-03-15:
It's my home wifi. I tried to deauth it later, before it stopped vorking, and it was success

tobozo, 2018-03-15:
you must select your AP first, then launch the attack
also everything is logged, so depending on how you initially flashed it, you may have to re-flash the deauther from time to time as things will eventually get overwritten and messed up :p

avanguard6, 2018-03-15:
I have reflashed my deauther and noticed that deauth works only for my phone and Ipad but on pc networks works without any difficulties. Another pc also works normally, but earlier the same computers lost access to the Internet after activation.
From time to time i see  "error loading attackInfo.json" on the top.
I really need to make it working on pcs

spacehuhn, 2018-03-24:
Not every device is vulnerable and you can loose connection when starting the attack. That's why you get the error. It says it on th website.

closed 2018-03-24 by spacehuhn