SpacehuhnTech/esp8266_deauther #734: Beacon thoughts

Raqriros, 2018-03-15:
Hi,
I'm sorry for hitting here but I wasn't sure how to properly make a pull request. 
Anyway I can delete this.

I just got an idea about beacon attack. It's clear you can acces to these fake 'wifi'. 
What about taking it a step further? 
Is this possible to make something like WiFi hotspot that asks you to Login?
And then this 'logging' will take us to page with for example gif with porn sounds on full volume?
What do you think about that? 
If its a good idea could someone maybe make this goddamn pull request?

spacehuhn, 2018-03-15:
Next version will make it easier to do that.

`wtf` label for the porn sounds 🤣

tobozo, 2018-03-16:
+1 for adding GOATSE visuals too, for the hearing impaired #a11y 

also this will require a sdcard reader as sounds + images probably won't fit in flash memory

Raqriros, 2018-03-17:
@spacehuhn  When are you planning version 2?
@tobozo  That's a good idea. I can imagine the awkwardness in that situation :D. Is this going to be hard to add sd card to esp?

tobozo, 2018-03-17:
probably easier than adding sound to GIF

Paffo, 2018-03-22:
This is exactly what was in my mind when i asked if he could implement Karma attack xD
But i was going to Rickroll them, not porn sound them.

closed 2018-03-24 by spacehuhn