SpacehuhnTech/esp8266_deauther #739: BUG New SDK

ChandraOrbit, 2018-03-24:
For the latest SDK I found BUG on Arduino IDE 1.8.5.
I have not tried any other version of IDE.
In the near future I will see whether the error is on my computer or from my IDE.

`Index error: could not find referenced tool name=esptool version=0.4.6 packager=esp8266
Index error: could not find referenced tool name=xtensa-lx106-elf-gcc version=1.20.0-26-gb404fb9-2 packager=esp8266
Index error: could not find referenced tool name=mkspiffs version=0.1.2 packager=esp8266`

[image]

spacehuhn, 2018-03-24:
Have you tried this already?

> Sometimes you can get a compiler error that can be fixed by also installing the original ESP8266 Arduino SDK. The steps are similar. Add http://arduino.esp8266.com/stable/package_esp8266com_index.json to the Boards Manager, search for "esp8266" and install the arduino-esp8266 (latest version should be fine). For a more detailed explanation visit github.com/esp8266/Arduino.

tobozo, 2018-03-24:
Install the regular esp8266 package first, it has all the tools your IDE is complaining about.

Those tools didn't exist in that version when SDK2.0.0 was released, and they haven't been added to avoid conflicts on deployment.

ChandraOrbit, 2018-03-24:
So far I use windows 7 ultimate 64 bit and it runs fine if using version 2.0.0.
A few moments ago I tried to use the new SDK and found the error.

What I do next:
1. Remove version 2.0.0
2. Install the latest version
3. Install the deauth SDK

The result still finds an error. I've done it 2 times. with the same result.

I'll try again

tobozo, 2018-03-24:
I'll try to replicate that, what version of the IDE are you using? Is that the one from the microsoft store?

ChandraOrbit, 2018-03-24:
I downloaded Arduino IDE from: https://www.arduino.cc/en/main/software
Last month when I flash my new board. which I bought on [Aliexpress](https://www.aliexpress.com/store/product/WiFi-deauther-OLED-V2-include-Case-and-Antenna/2996024_32825138910.html?spm=2114.12010612.0.0.5fae24e0Mt9pBm).

tobozo, 2018-03-24:
There are three types of install:
1) setup
2) zip
3) microsoft store

which one did you pick ?

ChandraOrbit, 2018-03-24:
Sorry, my English is bad.
I download it from Microsoft Store

tobozo, 2018-03-24:
Just tried this procedure and it worked flawlessly:

- download & clean install 
- add both esp8266 and esp8266_deauther boards in the prefs
- download the esp8266 latest SDK
- download the esp8266_deauther SDK
- run the example sketch

So the only option left is probably in the preferences of your IDE.
Please try to apply this solution: http://forum.arduino.cc/index.php?topic=490285.0

ChandraOrbit, 2018-03-24:
OK thanks, I will try to follow your instructions.

tobozo, 2018-03-24:
just reproduced the error by installing the SDKs in the wrong order, I'll update the documentation accordingly.

ChandraOrbit, 2018-03-25:
OK masalah BUG di SDK 2.0 selesai dengan mengikuti manual pada readme terbaru. Terima kasih.

closed 2018-03-25 by ChandraOrbit