SpacehuhnTech/esp8266_deauther #741: Crash on deauthing stations

spacehuhn, 2018-03-24:
So someone on my Discord server found this bug.
When you select stations and start deauthing, it can happen that the device crashes. Sometimes with a huge hex dump, sometimes without any message.

We're investigating this issue and hoping to fix it soon.

spacehuhn, 2018-03-25:
That should have fixed it 
https://github.com/spacehuhn/esp8266_deauther/commit/e494d95c15e689edd8eff140015c2fd53191d3fd

mlg556, 2018-03-26:
Is this fix included in the 2.0 version ? Or should I update it manually ? Because I am getting the same bug: the device crashes when deauthing selected stations with the following error:

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)
`
wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
`

spacehuhn, 2018-03-27:
Just download the current version again and flash it. I updated the code so it should be fine now.

closed 2018-04-11 by spacehuhn