SpacehuhnTech/esp8266_deauther #743: Automatic restart when attack starts in the Oled interface

vuong244, 2018-03-25:
When I attacked the client found in the oled interface, my board was restarted. I tried with another circuit board, the same result

tobozo, 2018-03-25:
ping #741.

How many stations did you select ?
Did you select the stations using the buttons, the web GUI, or from the terminal ?

Can you test the following situations:
- attack without selecting any stations
- attack when only one station is selected (and increment the number of selected stations)

Also can you specify the phone/browser you're using along with the version of the deauther.

spacehuhn, 2018-03-25:
Fixed with commit https://github.com/spacehuhn/esp8266_deauther/commit/e494d95c15e689edd8eff140015c2fd53191d3fd
So just download/clone the repository, compile and upload and you should be good. If you need a .bin file let me know but I strongly recomment using Arduino as explained here: https://github.com/spacehuhn/esp8266_deauther/commit/e494d95c15e689edd8eff140015c2fd53191d3fd 
Because of the better debugging possibilities.

closed 2018-03-25 by spacehuhn

vuong244, 2018-03-25:
Thank you. it was working