SpacehuhnTech/esp8266_deauther #744: charging current

Basti77, 2018-03-25:
Is the charging current intentionally set to 100mA?
How can I change this?
Loading a LiOn cell takes so long.

spacehuhn, 2018-03-25:
You're probably talking about the DSTIKE boards (?).
I don't think that hehy have a 100mA limit but I'm not sure. I rarely charge batteries over them.
Ping @lspoplove

Basti77, 2018-03-25:
JA genau. DSTIKE 
gehört die frage hier nicht hin?

tobozo, 2018-03-25:
Deauther issues belong here, this is more of a question related to a product specification, which the vendor (@lspoplove) can answer.

Phoenix1747, 2018-03-25:
I didn't observe the battery charging myself, but maybe your battery was overdischarged and needed to be 'revived' slowly.

Basti77, 2018-03-25:
I'm a model aviator and have a lot to do with batteries.
I have never experienced such behavior.
I have a test load on my JUNSI charger. the cell is OK.
The battery was only used on the Dstrike board and that has a deep discharge protection?

tobozo, 2018-03-25:
please post pics of your model and a link to the product page

Phoenix1747, 2018-03-25:
Strange, have you observed this behaviour for the whole time it took the cell to charge?

lspoplove, 2018-03-26:
Hello, I use IP5306 solution charging current is 2A.
Your battery maybe discharged too much or not avaliable again.
You have to wait when trickle charge process finish then it will reach 2A

Basti77, 2018-03-26:
My Hardware.
[image]
[image]

tobozo, 2018-03-26:
I have this model too and charge the battery from a separate charger, since you can't turn the deauther off while it's charging, the OLED stays on and the charge speed isn't ideal.

spacehuhn, 2018-04-04:
The version 2 board might not have a high current charging circuit, I also charge them with a seperate charger. The new deauther boards come with better charging chips.
I hope we answered all the questions.

Phoenix1747, 2018-04-05:
@spacehuhn Are there any detailed hardware specs available for the deauther boards?

tobozo, 2018-04-29:
closing this as solved

closed 2018-04-29 by tobozo

rayzzzor, 2024-05-02:
Hey everybody, I have the deauther Dstrike SE. I will charge it and every light is like it should be  but when I try to use it without charging it will not turn on anymore. Somebody can help pls? I got the watch yesterday
[image]