SpacehuhnTech/esp8266_deauther #745: [DE] German Language Support

spacehuhn, 2018-03-25:
With version 2.0, you can make a language file for the web interface.  
I made one for German as a test. You can change the language by going to the settings tab on the web interface and changing the `lang` setting from `en` to `de`.  

You can find the language files at [web_interface/lang](https://github.com/spacehuhn/esp8266_deauther/tree/master/web_interface/lang).  

I'm sure I have some typos and grammer/spelling mistakes in the de.lang as I quickly wrote it one night without much spell checking.  

Feel free to comment here if you wish any changes to the translation file or edit the file yourself and make a pull request.

tobozo, 2018-03-27:
There's also the [language.h](https://github.com/spacehuhn/esp8266_deauther/blob/master/esp8266_deauther/language.h) file although not everything should be translated there.

I can eventually provide a French translation for web UI and terminal, and use this opportunity to also suggest a way to contribute languages from outside language.h (i.e. by adding `extern` to `static const char` for translatable terms).

vuong244, 2018-04-07:
I tried changing "en" to "de", but the web interface language has not changed.

vuong244, 2018-04-07:
I tried changing "en" to "de", but the web interface language has not changed

spacehuhn, 2018-04-07:
Probably because your browser has the old language file cached and doesn't want to reload it.
Try deleting your browser cache.
Or open `192.168.4.1/lang/default.lang` and reload it one or two times.