SpacehuhnTech/esp8266_deauther #749: Fixed a type issue

michelep, 2018-03-26:
in file DisplayUI.h, line 159, i have to change "const char DejaVu_Sans_Mono_12[] PROGMEM = {
" to "const uint8_t DejaVu_Sans_Mono_12[] PROGMEM = {" to avoid compliation error:

invalid conversion from 'const char*' to 'const uint8_t* {aka const unsigned char*}' [-fpermissive]

Cheers !

tobozo, 2018-03-26:
Same for me, it started to happen after I accepted to update an Adafruit lib in the library manager.

Confirmed by enabling compile log and seeing this:
```
Multiple libraries were found for "SH1106Wire.h"
 Used: ~/Arduino/libraries/ESP8266_and_ESP32_Oled_Driver_for_SSD1306_display
```

Applying the same solution makes the warning go away and does not seem to affect the UI.

spacehuhn, 2018-03-26:
Weird, those are the default types when generating a font and using the library. 
Does it fix it when you change it from `const char` to `unsinged const char`?

tobozo, 2018-04-02:
commit [14b99ef](https://github.com/tobozo/Arduino/commit/14b99ef30200bc190518e0a019c78281ca6a0ca9) should have temporarily fixed that

closing this issue

closed 2018-04-02 by tobozo

vuong244, 2018-04-04:
Arduino: 1.8.3 (Windows 7), Board: "WEMOS D1 mini, 160 MHz, 80MHz, DIO, 115200, 4M (1M SPIFFS)"

Build options changed, rebuilding all
sketch\DisplayUI.cpp: In member function 'void DisplayUI::setupDisplay()':

DisplayUI.cpp:16: error: invalid conversion from 'const uint8_t* {aka const unsigned char*}' to 'const char*' [-fpermissive]

   display.setFont(DejaVu_Sans_Mono_12);

                                      ^

In file included from C:\Users\TRUONGAN\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\libraries\esp8266-oled-ssd1306-3.2.7/SH1106Wire.h:31:0,

                 from sketch\A_config.h:7,

                 from sketch\Settings.h:8,

                 from sketch\DisplayUI.h:10,

                 from sketch\DisplayUI.cpp:1:

C:\Users\TRUONGAN\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\libraries\esp8266-oled-ssd1306-3.2.7/OLEDDisplay.h:187:10: error:   initializing argument 1 of 'void OLEDDisplay::setFont(const char*)' [-fpermissive]

     void setFont(const char *fontData);

          ^

Multiple libraries were found for "Adafruit_NeoPixel.h"
 Used: C:\Program Files\Arduino\libraries\Adafruit_NeoPixel-master
 Not used: C:\Users\TRUONGAN\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\libraries\Adafruit_NeoPixel-1.1.6
Multiple libraries were found for "ArduinoJson.h"
 Used: C:\Users\TRUONGAN\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\libraries\ArduinoJson
 Not used: C:\Program Files\Arduino\libraries\ArduinoJson-master
Multiple libraries were found for "SH1106Wire.h"
 Used: C:\Users\TRUONGAN\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\libraries\esp8266-oled-ssd1306-3.2.7
 Not used: C:\Program Files\Arduino\libraries\esp8266-oled-ssd1306-master
 Not used: C:\Program Files\Arduino\libraries\esp8266-oled-ssd1306-master
 Not used: C:\Program Files\Arduino\libraries\esp8266-oled-ssd1306-master
 Not used: C:\Program Files\Arduino\libraries\esp8266-oled-ssd1306-master
Multiple libraries were found for "LinkedList.h"
 Used: C:\Users\TRUONGAN\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\libraries\LinkedList
 Not used: C:\Program Files\Arduino\libraries\LinkedList-master
exit status 1
invalid conversion from 'const uint8_t* {aka const unsigned char*}' to 'const char*' [-fpermissive]

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

tobozo, 2018-04-04:
@vuong244 the fix is in the first comment of this issue, otherwise delete and reinstall the SDK for esp8266deauther from the boards manager.

lehuukhoa, 2018-04-16:
@vuong244 bác xóa    display.setFont(DejaVu_Sans_Mono_12); là ok